openwrt/luci #275: luci does not support qmi modems

luizluca, 2015-01-03:
These modems are getting popular:

http://wiki.openwrt.org/doc/recipes/3gdongle#devices_using_qmi_and_mbim_protocol

wschlich, 2015-11-05:
I'd also love to have uqmi proto support in luci.

Currently, I have to use uci to configure qmi devices:

uci set network.wdm0=interface
uci set network.wdm0.proto='qmi'
uci set network.wdm0.device='/dev/cdc-wdm0'
uci set network.wdm0.pin='0000'
uci set network.wdm0.apn='internet.telekom'
uci set network.wdm0.username='t-mobile'
uci set network.wdm0.password='tm'
uci set network.wdm0.auth='pap'
uci set network.wdm0.modes='all'
uci set network.wdm0.delay='10'

uci set network.wwan=interface
uci set network.wwan.proto='dhcp'
uci set network.wwan.ifname='wwan0'

cron410, 2016-03-30:
The problem is not all modems respond to QMI commands the same way. I currently have a "Netgear" Sierra/Qualcomm 341u that responds to the QMI Disconnect command with "No Effect" and cannot be switched to 802.3 mode unless the modem is in a Disconnected state.

cshoredaniel, 2016-12-09:
@luizluca @cron410  @wschlich Can you help by testing PR#840

cron410, 2016-12-10:
Would I have to compile an image? Never been successful with that. I'm running openwrt x86.

luizluca, 2016-12-10:
@cshoredaniel  I'll try todo it on monday as I don't have a modem with me.
@cron410 , compiling luci would be enough but it's easier to just build it all. Not only compile luci, you would need to use this PR source as it is not merged yet.

cron410, 2016-12-10:
Yea probably won't have time this weekend for that. I'm terrible at
compiling things and adding source patches. More of a sysadmin making my
baby steps into dev.

cshoredaniel, 2016-12-13:
Hey no worries any time you have is good; I'm when I can too, although that is more than usual right now!

cshoredaniel, 2017-01-12:
QMI added with merge of https://github.com/openwrt/luci/pull/840

closed 2017-01-12 by cshoredaniel