openwrt/luci #280: Error when login.

CyberRSR, 2015-01-07:
/usr/lib/lua/luci/dispatcher.lua:448: Failed to execute function dispatcher target for entry '/'.
The called action terminated with an exception:
/usr/lib/lua/luci/sauth.lua:87: Session data invalid!
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:448: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:195: in function </usr/lib/lua/luci/dispatcher.lua:194>

tru7, 2015-01-18:
Stepped over the same problem today for trunk. It works fine with plain luci install including luci-theme-bootstrap. But as soon as luci-theme-openwrt or another theme is installed I get this exception. If I remove the extra theme it's working again. Is it the same for you?

hnyman, 2016-08-24:
Themes work for me. Session handling has been reworked since the this issues was filed. Likely fixed by now.

closed 2016-08-24 by hnyman