openwrt/luci #281: [luci-mod-admin-full] missing dependency after directory restructure

chris5560, 2015-01-08:
```
make[4]: Entering directory '/home/x/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/luci-mod-admin-full/src'
rm -f luci-bwc *.o
i486-openwrt-linux-uclibc-gcc  -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable  -I/home/x/trunk/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I/home/x/trunk/staging_dir/target-i386_i486_uClibc-0.9.33.2/include -I/home/x/trunk/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/x/trunk/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/include  -fpic -c -o luci-bwc.o luci-bwc.c
luci-bwc.c:35:20: fatal error: iwinfo.h: No such file or directory
 #include <iwinfo.h>
                    ^
compilation terminated.
Makefile:2: recipe for target 'luci-bwc.o' failed
make[4]: *** [luci-bwc.o] Error 1
make[4]: Leaving directory '/home/x/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/luci-mod-admin-full/src'
../../luci.mk:206: recipe for target '/home/x/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/luci-mod-admin-full/.built' failed
make[3]: *** [/home/x/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/luci-mod-admin-full/.built] Error 2
```

jow-, 2015-01-08:
Please see if the commit above fixes your problem.

chris5560, 2015-01-08:
@jow- 
I think I missed one patch of admin module.
In my Makefile was no PKG_BUILD_DEPENDS:= definition.
I have to check both versions after "make dirclean" because after this it happen the first time.
By the way, I think that both admin modules depends on libiwinfo-lua to be also installed.

chris5560, 2015-01-08:
@jow- 
I checked both, each after "make dirclean"
PKG_BUILD_DEPENDS:=+libiwinfo => compiles without errors
PKG_BUILD_DEPENDS:=iwinfo => compiles also without errors
Is there a "+" needed in the second version ?
Christian

closed 2015-01-08 by chris5560