openwrt/luci #283: Missing Lua, Transtaltion?

hitech95, 2015-01-09:
Since yesterday, LIGHTS seems to have problems.
Initially I could not fill OpenWRT but today LUCI interface is not "accessible".

Unable to launch the requested CGI program:
   / www / cgi-bin / lights: No such file or directory

Seems to have been lifted from the dependencies Lua. (I do not know if it is correct)

I also do not understand why the languages are no longer modular, I can not set the flag "m".

hitech95, 2015-01-10:
I saw that was added the dependency.
I think it is more appropriate that LUCI add directly addiction.

Now you must first select LUA, then LUCI.
Also if LUA is not selected, you can select the translations and app modules. Instead you can not select the collection.

hnyman, 2015-08-29:
Dependency has been fixed by https://github.com/openwrt/luci/commit/3afecbd8abcdbc97e7f6f140517132db7e9639e3 and https://github.com/openwrt/luci/commit/b2ac7ad15a0d852b6b4ccae554db9dfea45c0f9e in January.

closed 2015-08-29 by hnyman