openwrt/luci #293: refer to Luci wiki ,i type “make runhttpd",but there is No rule to make target `runhttpd'

shindowarch, 2015-01-16:
i download the luci-0.12 package,and unzip the package,
then
cd luci
make
but there is No rule to make target `runhttpd'

kumarrishav, 2015-03-10:
@jow-  any update on this or any other way to build and run Luci?

jow-, 2015-03-10:
I suggest to build LuCI as part of OpenWrt and then use an SSHFS mount or an SFTP capable editor to edit LuCI directly on the target system.

The environment required to run LuCI (uhttpd, ubus, libubox, rpcd, uci ...) became too huge so its not feasible to simulate it on a desktop anymore.

errorcode7, 2015-04-24:
the makefile has a target `runuhttpd' ,but wen i run make there is no config.mk file in build dir,s there anything wrong with me?
i just want to run the luci in my archlinux,is that possible?

cshoredaniel, 2016-12-10:
runuhttpd no longer supported.

closed 2016-12-10 by cshoredaniel