openwrt/luci #298: theme-bootstrap: no space between tab-description and first value

chris5560, 2015-01-18:
If using a tab description there is not enough space left to the first value.
Have a look on attached picture. Tab description touch start button.

[image]

legendtang, 2015-02-28:
The latest `luci-app-privoxy` package added three `<br>` beneath the text. It should works well now.
[image]

This issue should close now.

chris5560, 2015-02-28:
Hi Legend,
I opened this issue because it's only a problem of bootstrap-theme, so of bootstrap.css.
I'm not familiar with css, so I can't fix it.
Inside luci-app-privoxy and luci-app-ddns I build a workaround by inserting three <br />.
It's a workaround not a fix.

legendtang, 2015-02-28:
Hello chris5560,
Sorry for the misinterpretation. I think I should be able to fix them with a pull request.
That means I should remove those `<br />` in these two luci-apps at the same time. Is that right?

chris5560, 2015-02-28:
Legend,
if you could fix the bootstrap.css I will modify both apps.
I had some other minor changes for those apps pending for distribution.
So all for those apps could be done at once.
Thanks
Christian

legendtang, 2015-02-28:
Hi Christian,
I have fixed just now. It's because the `form .cbi-tab-descr` has no definition in CSS.
Please remove those `br` as soon as possible.

Thanks
Legend Tang

chris5560, 2015-03-01:
Fixed. Thanks.

closed 2015-03-01 by chris5560