openwrt/luci #303: Using the latest version of the DDNS error

qiuzi, 2015-01-20:
[image]
[image]

Configure a single service display normal：
[image]

chris5560, 2015-01-20:
I cannot follow your problem. Description missing.
Hopefully details can be found inside logfile at /var/log/ddns/[sectionname].log
Please send one to me by direct mail (have a look into header of scripts in /usr/lib/ddns)
Please also provide versions of ddns-scripts and luci-app-ddns. (Click on blue "Dynamic DNS" header on overview page).

qiuzi, 2015-01-21:
How much of your mail?

Luci-app-ddns=2.1.1-0

Ddns-scripts=2.1.0

chris5560, 2015-03-19:
@qiuzi
is it fixed with the current version of ddns-scripts ?

chris5560, 2015-04-12:
Newer versions of ddns-scripts and luci-app-ddns available. Please update and create new issue if problem still exists.

closed 2015-04-12 by chris5560