openwrt/luci #309: "Bad Gateway" and segfault in ip.so

chris5560, 2015-01-25:
git-15.025.63007-5317b09 / r44138
```
kern.info kernel: [  131.275074] luci[2097]: segfault at 9003030 ip b7622b0f sp bfc78880 error 4 in ip.so[b7621000+5000]
```
Running luci-app-ddns. pressing the [edit] button will show "bad gateway" and syslog show the segfault.
I don't know exactly which change introduce this error. There was no change in the code of overview.lua. The calling code:
 ```
local DISP = require "luci.dispatcher"
local HTTP = require "luci.http"
...
ts = m:section( TypedSection, "service",
	translate("Overview"),
	translate("Below is a list of configured DDNS configurations and their current state." .. "<br />" ..
		"If you want to send updates for IPv4 and IPv6 you need to define two separate Configurations " ..
		"i.e. 'myddns_ipv4' and 'myddns_ipv6'") )
ts.sectionhead = translate("Configuration")
ts.template = "cbi/tblsection"
ts.addremove = true
ts.extedit = DISP.build_url("admin", "services", "ddns", "detail", "%s")
function ts.create(self, name)
	AbstractSection.create(self, name)
	HTTP.redirect( self.extedit:format(name) )
end
```

chris5560, 2015-01-25:
Supporting currently a user running git-15.024.48411-9feb1b8 without problems.

jow-, 2015-01-25:
I think the issue is fixed after 0d32e69f163e80442c7a483c445484ee250986e8

chris5560, 2015-01-25:
Fixed

closed 2015-01-25 by chris5560