openwrt/luci #322: XML parsing error on multiple tabs

hg0noz, 2015-02-08:
I got the following error:
XML Parsing Error: not well-formed Location: http://192.168.2.1/cgi-bin/luci/;stok=b13d872438c4adaf0d8c80c655d7f515/admin/services/ Line Number 183, Column 57:	/usr/lib/lua/luci/dispatcher.lua:195: in function </usr/lib/lua/luci/dispatcher.lua:194></pre> ---------------------------------------------------------------^

On the overview section Network, Wireless, DHCP sections does not loading either.

Router Model	TP-Link TL-WR2543N/ND v1
Firmware Version	OpenWrt Attitude Adjustment 12.09 / LuCI Trunk (trunk+svn8893)
Kerner Version      3.3.8

jow-, 2015-02-08:
This version is really old, please try OpenWrt 14.07 or later.

closed 2015-02-08 by jow-

hg0noz, 2015-02-08:
Just refreshed to 14.07. Running checks.

hg0noz, 2015-02-08:
Now with OpenWrt Barrier Breaker 14.07 / LuCI Trunk (trunk+svn8893) and the issue is still exists.

jow-, 2015-02-08:
The I need more details. If you open the source view in your browser you should see a stacktrace somewhere, indicating the exact error.

hg0noz, 2015-02-08:
Here it is:
<h2><a id="content" name="content">500 Internal Server Error</a></h2>
<p>Sorry, the server encountered an unexpected error.</p>
<pre class="error500">/usr/lib/lua/luci/dispatcher.lua:449: Failed to execute firstchild dispatcher target for entry '/admin/services'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:449: Failed to execute cbi dispatcher target for entry '/admin/services/p910nd'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:105: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:105: Failed to execute template 'cbi/tsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:105: Failed to execute template 'cbi/ucisection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:105: Failed to execute template 'cbi/network_netlist'.
A runtime error occured: /usr/lib/lua/luci/model/cbi/p910nd.lua:40: attempt to call method 'get_status_by_address' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:449: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:195: in function </usr/lib/lua/luci/dispatcher.lua:194></pre>
<div class="clear"></div>
</div>
</div>

<p class="luci">Powered by LuCI Trunk (trunk+svn8893)</p>
</body>
</html>

hg0noz, 2015-02-11:
I have resolved the issue by force reinstall luci.