openwrt/luci #332: Wrong WAN connection on status page

r3pek, 2015-02-25:
I have two connections (on different VLANs) that are made from the same physical connection. One is the WAN interface, and the other is the IPTV interface. The main status page wrongly picks the IPTV interface as the "IPv4 WAN status". Everything still works, it's just that the information is wrong (even the IP range is not on the public addressable class)

DennisEdlund, 2018-05-08:
This is unlogical, the LAN interface won't become WAN interface just because you preset the gateway to the CORRECT value.
You still need to set ALL the other parameters, like the custom DNS and broadcast (not shure you need to set broadcast).
Why do we need to set DNS if we want to use the default given from the gateway, and the broadcast is given by the preset mask?
So it's unlogical NOT to set the gateway parameter when you need to set DNS.

jow-, 2018-06-25:
Current master refers to default-route interfaces as "Upstream" now to prevent confusion with wan/lan labelling.

closed 2018-06-25 by jow-