openwrt/luci #335: luci-app-ddns: error in system log every five seconds on Overview page

hnyman, 2015-02-28:
luci-app-ddns causes errors to system log every five seconds if the Overview front page is open (as the status items are refreshed every five seconds). The recent changes in uhttpd error logging has probably surfaced this now. 

I have already filed a bug on Openwrt bug tracker ( https://dev.openwrt.org/ticket/19083 ), but as this error seems to be caused by the version check function of this app, so I am filing also this issue ticket here.

I did not try debugging more exactly if this is coming from ipkg_ver_installed or from ipkg_ver_compare:
https://github.com/openwrt/luci/blob/819a733df3c9e93619dbad1159f67de70d406da7/applications/luci-app-ddns/luasrc/tools/ddns.lua#L99

errors in log  (there is also a similar iptables error from the upnp app):
```
Sat Feb 28 19:51:37 2015 daemon.err uhttpd[27651]: Collected errors:
Sat Feb 28 19:51:37 2015 daemon.err uhttpd[27651]:  * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
Sat Feb 28 19:51:37 2015 daemon.err uhttpd[27651]: Collected errors:
Sat Feb 28 19:51:37 2015 daemon.err uhttpd[27651]:  * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
Sat Feb 28 19:51:37 2015 daemon.err uhttpd[27651]: iptables: No chain/target/match by that name.
Sat Feb 28 19:51:42 2015 daemon.err uhttpd[27651]: Collected errors:
Sat Feb 28 19:51:42 2015 daemon.err uhttpd[27651]:  * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
Sat Feb 28 19:51:42 2015 daemon.err uhttpd[27651]: Collected errors:
Sat Feb 28 19:51:42 2015 daemon.err uhttpd[27651]:  * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
Sat Feb 28 19:51:42 2015 daemon.err uhttpd[27651]: iptables: No chain/target/match by that name.
Sat Feb 28 19:51:47 2015 daemon.err uhttpd[27651]: Collected errors:
Sat Feb 28 19:51:47 2015 daemon.err uhttpd[27651]:  * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
Sat Feb 28 19:51:47 2015 daemon.err uhttpd[27651]: Collected errors:
Sat Feb 28 19:51:47 2015 daemon.err uhttpd[27651]:  * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
Sat Feb 28 19:51:47 2015 daemon.err uhttpd[27651]: iptables: No chain/target/match by that name.
Sat Feb 28 19:51:52 2015 daemon.err uhttpd[27651]: Collected errors:
Sat Feb 28 19:51:52 2015 daemon.err uhttpd[27651]:  * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
Sat Feb 28 19:51:52 2015 daemon.err uhttpd[27651]: Collected errors:
Sat Feb 28 19:51:52 2015 daemon.err uhttpd[27651]:  * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
Sat Feb 28 19:51:52 2015 daemon.err uhttpd[27651]: iptables: No chain/target/match by that name.
Sat Feb 28 19:51:57 2015 daemon.err uhttpd[27651]: Collected errors:
Sat Feb 28 19:51:57 2015 daemon.err uhttpd[27651]:  * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
Sat Feb 28 19:51:57 2015 daemon.err uhttpd[27651]: Collected errors:
Sat Feb 28 19:51:57 2015 daemon.err uhttpd[27651]:  * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
```

@chris5560

hnyman, 2015-02-28:
Something similar should probably be done that Jow did for luci-app-upnp by https://github.com/openwrt/luci/commit/cf4b89b06be6d6191965e18517a39f870aa5e729 i.e. redirecting opkg's stderr to /dev/null

 
It also seems rather wasteful to do the version comparison at every update of the Overview page, every five seconds. Maybe there should be logic to do the check just once

hnyman, 2015-02-28:
Adding stderr redirect to /tools/ddns.lua seems to help:
```
root@OpenWrt:/usr/lib/lua/luci/tools# diff /rom/usr/lib/lua/luci/tools/ddns.lua ddns.lua
--- /rom/usr/lib/lua/luci/tools/ddns.lua
+++ ddns.lua
@@ -114,7 +114,7 @@
                return nil
        end
        -- opkg list-installed [pkg] | cut -d " " -f 3 - return version as sting
-       local ver = SYS.exec([[opkg list-installed ]] .. pkg .. [[ | cut -d " " -f 3 ]])
+       local ver = SYS.exec([[opkg list-installed ]] .. pkg .. [[ 2>/dev/null | cut -d " " -f 3 ]])
        if (#ver > 0) then
                return ver
        end

```

jow-, 2015-02-28:
A less resource-intensive way is reading the package control file directly (and doing the file content parsing in Lua instead of relying on `cut` etc.)

    #!/usr/bin/lua
    
    require "luci.util"
    
    function ipkg_ver_installed(pkg)
    	local version = nil
    	local control = io.open("/usr/lib/opkg/info/%s.control" % pkg, "r")
    	if control then
    		local ln
    		repeat
    			ln = control:read("*l")
    			if ln and ln:match("^Version: ") then
    				version = ln:gsub("^Version: ", "")
    				break
    			end
    		until not ln
    		control:close()
    	end
    	return version
    end
    
    v = ipkg_ver_installed("ddns-scripts")
    print(v)

The advantage of this approach is that it requires no external forks and that it works as unprivileged users. The `opkg` command will fail when run as non-root due to its inability to acquire a database lock.

chris5560, 2015-03-01:
Many Thanks. Will fix it a.s.a.p.

closed 2015-03-19 by chris5560