openwrt/luci #342: httpclients request_raw crashed if sock:readall() return nil

paalsteek, 2015-03-03:
In my usecase sock:readall() in libs/luci-lib-httpclient/luasrc/httpclient.lua:336 returns nil which causes a crash as concatenation of nil values is not allowed.
This bug is caused by c30cf9abd4e1e46731e36552ae2f459a27c5cf46 and can be fixed with the following patch:

```patch
--- a/libs/luci-lib-httpclient/luasrc/httpclient.lua
+++ b/libs/luci-lib-httpclient/luasrc/httpclient.lua
@@ -333,7 +387,7 @@ function request_raw(uri, options)
 		end
 	end
 	
-	return response.code, response, linesrc(true)..sock:readall(), sock
+	return response.code, response, linesrc(true)..(sock:readall() or ""), sock
 end
 
 function cookie_parse(cookiestr)
```

cshoredaniel, 2016-12-10:
This appears to have been fixed in current LuCI; can you verify?

cshoredaniel, 2017-01-17:
No response in a month; appears to have been fixed in trunk.

closed 2017-01-17 by cshoredaniel