openwrt/luci #346: ddns-scripts wront outdated warning when using no-ip without no-ip support

Zokormazo, 2015-03-12:
If I build ddns-scripts without no-ip support and add a no-ip ddns host to dynamic-dns, Status-> Overview complains saying that I have an outdated ddns-scripts package.

Shouldn't warning something different for this case? The package is not outdated, it lacks no-ip support.

chris5560, 2015-03-13:
It looks like that something went wrong during update from older version.
The current system_status.htm has no js code to display something about old version.
On ddns service->overview there is still a version check done and displayes under "Hints".
When you click on the link "Dynamic DNS" on ddns overview page you will get version information for ddns-scripts and luci-app-ddns.

chris5560, 2015-03-31:
@Zokormazo 
Is the problem fixed or not ?
Thanks for feedback.
Christian

Zokormazo, 2015-04-03:
@chris5560 

As you said it was a some kind of garbage problem. I tried with a clean buildroot and reseted router and the problem is gone.

You can close it ;)

chris5560, 2015-04-03:
Thanks for feedback

closed 2015-04-03 by chris5560