openwrt/luci #350: basics.lua:17: attempt to call a nil value

JoeSem, 2015-03-25:
/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute cbi dispatcher target for entry '/admin/freifunk/basics'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/freifunk/basics.lua:17: attempt to call a nil value
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>

profile_xx is abaliable
All other functions are working well. Powered by LuCI (git-15.079.29361-3e37216)

hnyman, 2015-08-29:
Has been probably fixed by https://github.com/openwrt/luci/commit/5ef1485133ad0776cdfe028bc233641553e074f1 in May. 
Does it work now?

chris5560, 2015-11-15:
Closed because no feedback for month.

closed 2015-11-15 by chris5560