openwrt/luci #353: LuCI support for IP Set

JoeSem, 2015-03-27:
UCI Firewall V3 is supporting IP Set to simplify matching of huge address or port lists without the need for creating one rule per item to match.

Would be great to have this functionality also avaliable li LuCI Firewall.
For more details refer to http://wiki.openwrt.org/doc/uci/firewall#ip_sets

jhuebner79, 2017-01-15:
+1 - much anticipated feature

guidosarducci, 2018-03-22:
Let me also chime in that support for both "IP Sets" and "DSCP Marking" are my top wishes for `luci-app-firewall`, and I suspect for others too.

hnyman, 2023-01-14:
Closing this old feature request, as ipset support for firewall is currently being implemented in PR #6187 

(and for DCSP marking there is already the qosify package, although it is not supported in LuCI).

closed 2023-01-14 by hnyman