openwrt/luci #354: No web interface

wifiextender, 2015-03-28:
Hello openwrt,

I've purchased TP-LINK TL-WA701ND wifiextender couple weeks ago and today replaced it's firmware with yours.

Everything is smoothly so far, but I'd like to configure the device from my browser and saw that there was package called 'luci'.

Installed the non-ssl version of luci and when I went to the local ip address of my wifiextender saw that the web interface was kinda weird.

Attaching picture so you can see how weird the web interface is:

[image]

Cheers

thess, 2015-03-28:
Did you have any errors/warnings while installing luci??

Try: opkg list-installed luci* 

You should see several packages. Did you change anything in the uhttpd config?
You seem to be missing a number of files/directories in /www.

Others here may have a better answer.

wifiextender, 2015-03-30:
Hello Ted,

I've reset the device and removed luci with it is dependencies and installed it again. This time I paid more attention and it seems that my device is running out of space, here is the output:

```
opkg install luci
Installing luci (git-15.079.29361-3e37216-1) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/luci/luci_git-15.079.29361-3e37216-1_all.ipk.
Installing uhttpd (2015-03-11-ae3fe30b6a1d4a45a34b3a896e9a4997c5da500f) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/base/uhttpd_2015-03-11-ae3fe30b6a1d4a45a34b3a896e9a4997c5da500f_ar71xx.ipk.
Installing luci-mod-admin-full (git-15.079.29361-3e37216-1) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/luci/luci-mod-admin-full_git-15.079.29361-3e37216-1_ar71xx.ipk.
Installing luci-base (git-15.079.29361-3e37216-1) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/luci/luci-base_git-15.079.29361-3e37216-1_ar71xx.ipk.
Configuring luci-mod-admin-full.
Configuring uhttpd.
ln: /etc/rc.d/S50uhttpd: No space left on device
Collected errors:
 * wfopen: /usr/lib/lua/luci/http.lua: No space left on device.
 * extract_archive: Cannot make dir /usr/lib/lua/luci/sys/: No space left on device.
 * extract_archive: Cannot make dir /usr/lib/lua/luci/sys/zoneinfo/: No such file or directory.
 * wfopen: /usr/lib/lua/luci/sys/zoneinfo/tzdata.lua: No such file or directory.
 * wfopen: /usr/lib/lua/luci/sys/zoneinfo/tzoffset.lua: No such file or directory.
 * wfopen: /usr/lib/lua/luci/sys/zoneinfo.lua: No such file or directory.
 * wfopen: /usr/lib/lua/luci/sys/iptparser.lua: No such file or directory.
 * extract_archive: Cannot make dir /usr/lib/lua/luci/cbi/: No space left on device.
 * wfopen: /usr/lib/lua/luci/cbi/datatypes.lua: No such file or directory.
 * extract_archive: Cannot make dir /usr/lib/lua/luci/http/: No space left on device.
 * wfopen: /usr/lib/lua/luci/http/protocol.lua: No such file or directory.
 * extract_archive: Cannot make dir /usr/lib/lua/luci/http/protocol/: No such file or directory.
 * wfopen: /usr/lib/lua/luci/http/protocol/mime.lua: No such file or directory.
 * wfopen: /usr/lib/lua/luci/http/protocol/date.lua: No such file or directory.
 * wfopen: /usr/lib/lua/luci/http/protocol/conditionals.lua: No such file or directory.
 * wfopen: /usr/lib/lua/luci/i18n.lua: No space left on device.
 * wfopen: /usr/lib/lua/luci/cacheloader.lua: No space left on device.
 * wfopen: /usr/lib/lua/luci/controller/admin/servicectl.lua: No space left on device.
 * wfopen: /sbin/luci-reload: No space left on device.
 * extract_archive: Cannot make dir /www/luci-static/resources/icons/: No space left on device.
 * wfopen: /www/luci-static/resources/icons/ethernet.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/encryption_disabled.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/signal-none.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/wifi_disabled.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/ethernet_disabled.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/wifi.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/signal-0-25.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/vlan.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/switch_disabled.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/bridge_disabled.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/signal-25-50.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/port_down.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/signal-50-75.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/tunnel.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/signal-0.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/switch.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/signal-75-100.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/loading.gif: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/vlan_disabled.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/wifi_big.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/bridge.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/port_up.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/encryption.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/wifi_big_disabled.png: No such file or directory.
 * wfopen: /www/luci-static/resources/icons/tunnel_disabled.png: No such file or directory.
 * wfopen: /www/luci-static/resources/xhr.js: No space left on device.
 * wfopen: /www/luci-static/resources/cbi.js: No space left on device.
 * extract_archive: Cannot make dir /www/luci-static/resources/cbi/: No space left on device.
 * wfopen: /www/luci-static/resources/cbi/file.gif: No such file or directory.
 * wfopen: /www/luci-static/resources/cbi/arrow.gif: No such file or directory.
 * wfopen: /www/luci-static/resources/cbi/user.gif: No such file or directory.
[... cut ...]
 * wfopen: /lib/uci/upload/.gitignore: No such file or directory.
 * wfopen: /etc/config/luci: No space left on device.
 * wfopen: /etc/config/ucitrack: No space left on device.
 * pkg_write_filelist: Failed to open //usr/lib/opkg/info/luci-base.list: No space left on device.
 * opkg_install_pkg: Failed to extract data files for luci-base. Package debris may remain!
 * opkg_install_cmd: Cannot install package luci.
 * opkg_conf_write_status_files: Can't open status file //usr/lib/opkg/status: No space left on device.

```

closed 2015-03-30 by wifiextender

wifiextender, 2015-03-30:
What's even more interesting is that my **tmpfs** directory has 99% of the needed free space, and no matter what packages I am trying to install, the package manager will always output "no space left"

```root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
rootfs                  768.0K    696.0K     72.0K  91% /
/dev/root                 2.0M      2.0M         0 100% /rom
tmpfs                    13.9M    560.0K     13.4M   4% /tmp
/dev/mtdblock3          768.0K    696.0K     72.0K  91% /overlay
overlayfs:/overlay      768.0K    696.0K     72.0K  91% /
tmpfs                   512.0K         0    512.0K   0% /dev
```

Is there any package that is less than 70kb to delete and recreate my partitions ?