openwrt/luci #387: luci-app-ddns trunk error

debiansid, 2015-05-21:
Hints
Software update required
The currently installed 'ddns-scripts' package did not support all available settings.
Please update to the current version!

chris5560, 2015-05-21:
@debiansid 
Which version of ddns-scripts and luci-app-ddns are you running ?
Which available setting is missing ?
Christian

db260179, 2015-05-23:
The change of min version in the luci-app-ddns seems to be the cause of this

+DDNS_MIN = "2.4.2-1" -- minimum version of service required

Previously
-DDNS_MIN = "2.2.0-1" -- minimum version of service required
Doesnt show this error

debiansid, 2015-05-23:
@chris5560 
ddns-scripts - 2.4.2-1 
ddns-scripts_nsupdate - 2.4.2-1 - 
luci-app-ddns - 2.2.2-1

chris5560, 2015-05-25:
Fixed

closed 2015-05-25 by chris5560

debiansid, 2015-05-25:
2.4.2-1did not fix it here.

debiansid, 2015-05-25:
luci-app-ddns  2.2.4 I mean

debiansid, 2015-05-25:
Sorry,reboot openwrt show it correctly finally