openwrt/luci #392: Redirect to authorization on each action

falstaff84, 2015-05-24:
I'm using 15.05-rc1 on a Netgear WNDR3700v2. After using the web interface one day, I had issues the next day: I kept getting redirected to the login page on each action. After successful login, I'm trying to access a specific page (e.g. Network=> Interfaces), which redirects me on the login page. Explicitly loging out and log back in did not help. What helped is using a different browser (Firefox in my case) and using a complete new instance (closing Chromium completly and reopen). It seems that a long standing connection/browser session leads to session problems.

Note that I moved the laptop from Ethernet to WiFi connection, and also restarted the router. A logout is of course to be expected (after all, the IP address I'm accessing the router from changed). However, after login, the web interface should kept logged in...

Browser: Chromium on Linux, Version 42.0.2311.135 (64-bit)

luci - git-15.126.50380-7a54785-1
luci-app-firewall - git-15.126.50380-7a54785-1
luci-base - git-15.126.50380-7a54785-1
luci-lib-ip - git-15.126.50380-7a54785-1
luci-lib-nixio - git-15.126.50380-7a54785-1
luci-mod-admin-full - git-15.126.50380-7a54785-1
luci-proto-ipv6 - git-15.126.50380-7a54785-1
luci-proto-ppp - git-15.126.50380-7a54785-1
luci-theme-bootstrap - git-15.126.50380-7a54785-1

oneru, 2015-05-28:
@falstaff84 I get this when I have luci open in more than one tab. Apparently the auto-refresh from one tab is logging out the other. Not sure that was what was going on in your case.

falstaff84, 2015-05-28:
Ah yes, that could have been the case. And it worked while the session was valid, but as soon as it was invalid, it invalidated the just newly created session when I logged in.

However, since that could happen quite easily, that case should be handled better, imho..?

wpietri, 2015-09-29:
Wow, I just spent 10 minutes struggling with getting logged out at random. Totally confusing. It'd be great to see this handled more gracefully. If any window is going to to be forced to log out, it should be my old one that I've forgotten about, not the new one I'm trying to use to configure my router.

Randy000, 2015-12-03:
Your need rpcd.
run:
opkg update
opkg install rpcd

falstaff84, 2015-12-14:
@Randy000 I do have rpcd installed, I still can reproduce the issue: Open a browser tab, log in, stay idle on the front page (don't know how long the session timeout is, I left it open for an hour or so), open a second window, and try to use the web interface... You will get logged out within seconds...

Randy000, 2015-12-14:
@falstaff84  Please make sure you have installed several other packages.You can run.
opkg update
opkg install liblua
opkg install libuci-lua
opkg install rpcd
opkg install luci-lib-ip
opkg install libubus-lua
opkg install libiwinfo
opkg install libiwinfo-lua
opkg install luci-theme-bootstrap
opkg install luci-app-firewall
opkg install lua
opkg install luci-proto-ppp
opkg install luci-lib-nixio
opkg install luci-base
opkg install luci-mod-admin-full
opkg install luci-proto-ipv6
opkg install uhttpd
opkg install uhttpd-mod-ubus
opkg install luci

Kenterfie, 2016-03-18:
i have the same problem

HOSTED-POWER, 2016-04-04:
Hi,

Using 15.05.1 and I have the same issue!! Very annoying. I think it has to do with cookies because using chrome incognito works ... (After a while it stops, but re-opening chrome in incognito fixes it)

Very very annoying! :(

jpereira, 2016-11-01:
@HOSTED-POWER Please, check the rpc service.

```
root@router:/# /etc/init.d/rpcd enable
root@router:/# /etc/init.d/rpcd restart
root@router:/# 
```

HOSTED-POWER, 2016-11-02:
Hi,

It's running as far as I can see. Not sure if the issue still exists, didn't use luci for long time now :)

oemunoz, 2016-11-16:
Same here:
CHAOS CALMER (15.05.1, r48532)

opkg list | grep luci
luci - git-16.018.33482-3201903-1
....

uname -a
Linux oems 3.18.23 #1 Sun Jan 31 18:39:35 CET 2016 mips GNU/Linux

Restart rpcd dont work for me... 

But close all browser and open it in a different one, works... I dont now why.

cshoredaniel, 2016-12-10:
This appears to work properly in current LuCI trunk.

cshoredaniel, 2016-12-10:
This is discussed also in #858 so marking as dupe of that; it's been solved for trunk and it's not clear that it's reasonable to fix in 15.05, however that's to be decided.

closed 2016-12-10 by cshoredaniel