openwrt/luci #404: Creating two interfaces in same vlan and deleting one deletes both

txomon, 2015-06-09:
I can't really be sure about this, because I totally lost access to the device and had to flash via serial, but these are the steps and what I was doing.

I use openwrt in a router for vlan tagging, I have the WAN interface tag all the traffic with 802.1q. In order to manage the port tagging, I have a management interface (MNG) in one of those VLANs.

I wanted to change from vlan 40 to vlan 50, and in order not to loose access to the device, and maintain the same interface:
- created another interface (TMP) in the vlan 50, with an static IP.
- accessed through the TMP interface to the luci config
- moved the inteface MNG to vlan 50 with another static IP in the same net (and vlan)
- accessed through the MNG interface to the Luci dashboard
- removed TMP interface
- lost access to either of both interfaces

I also noticed that in the interface overview, TMP and MNG had the same ips, which makes me suspect that though they appeared split in the interface, when I clicked to remove TMP, it deleted both interfaces.

cshoredaniel, 2016-12-10:
Did you try rebooting after waiting a few minutes (to ensure config was written)?  It's possible the interface just went down and came back up.  Having multiple ip addresses to the same interfaces (e.g. eth1.50) works fine for me, and delete one interface only deletes that interface.  I'm wondering also whether you were using static IP addresses on the client or DHCP.

txomon, 2016-12-10:
I am afraid that it has been so long that I no longer remember what happened, or whether this situation is reproducible any more...

I will close the issue

closed 2016-12-10 by txomon