openwrt/luci #420: How to create "tabbed map" with cbi

chris5560, 2015-06-16:
It is easy to add tabs to a section, but I need a way to add tabs to a map.
A map access the whole config file and the config has multiple (named) sections.
I want to show each section in a separate tab below the map title/description.
Thanks for support
Christian

cshoredaniel, 2016-12-10:
Can you describe a little bit more what you mean?  Are you thinking of something like instead of having having a top down lists of section, of having the section names as tab headers and when you click you see the section?

thess, 2016-12-10:
@cshoredaniel  - your interpretation is correct. I think what @chris5560  is looking for is similar to something I wanted to do also (same time-frame). luci-app-mjpg-streamer would use this for multiple camera support which is supported by the uci config settings. Each tab represents particular camera settings and a shapshot while the top section is for the server settings.

chris5560, 2016-12-10:
I think we can close this Issue.
@jow- published required modification inside cbi someday in 2015
I use the property .tabbed = true inside Map definition inside luci-app-radicale.
It define one tab per section.
From my knowledge it's documented nowhere.
@cshoredaniel  I forgot to close this issue.
Sorry

closed 2016-12-10 by chris5560