openwrt/luci #427: RPC Authentication returns an "invalid params"

hertog6, 2015-07-14:
Hello,

I am trying to authenticate my client on the JSON-RPC API using the following method, but I get an error back.

Request;
```
curl -i -X POST -d '{"jsonrpc": "2.0", "method": "login", "params": { "user": "root", "password":"xxxx"}, "id": 1}' http://10.13.37.1/cgi-bin/luci/rpc/auth
```

Response;
```
HTTP/1.1 200 OK
Connection: close
Transfer-Encoding: chunked
Content-Type: application/json
Cache-Control: no-cache
Expires: 0

{"id":1,"error":{"message":"Invalid params.","data":"/usr/lib/lua/luci/controller/rpc.lua:53: attempt to index global 'luci' (a nil value)","code":-32602},"jsonrpc":"2.0"}
```

I based my request on a comment made in this topic; https://forum.openwrt.org/viewtopic.php?pid=211763#p211763

What did I do wrong?

hertog6, 2015-07-15:
I fixed it, so this issue can be closed! The error was partially my fault and bug in the RPC module.

The request should have been
```
{
    "jsonrpc": "2.0",
    "method": "login",
    "params": {
        "username": "root", 
        "password": "xxx"
    },
    "id": 0
}
```

And the bug in the RPC module I fixed by setting the timeout to a fixed value of 3600.

Before:
```
local sdat = util.ubus("session", "create", { timeout = luci.config.sauth.sessiontime })
```

After:
```
local sdat = util.ubus("session", "create", { timeout = 3600 })
```

jow-, 2015-07-15:
I've seen your message in IRC but you already left when I was about to respond. I believe the issue is fixed with 16047d03c54df4474cb675a9ce98d06aad5a656b

closed 2015-07-15 by jow-

hertog6, 2015-07-15:
Even tho this issue is closed, I just wanted to let you know that I tried this fix and it didn't fix the problem.

fbradyirl, 2015-07-18:
Same here. Still getting the same error.

(Using Python shell)
```
>>> res.text
'{"id":1,"error":{"message":"Invalid params.","data":"/usr/lib/lua/luci/controller/rpc.lua:55: attempt to index global \'luci\' (a nil value)","code":-32602},"jsonrpc":"2.0"}'
>>> res.request.body
'{"method": "login", "id": 1, "jsonrpc": "2.0", "params": ["root", "passw"]}'
>>> 
```

TDWhbp, 2015-09-05:
I too am having this issue... 
{"id":1,"error":{"message":"Invalid params.","data":"/usr/lib/lua/luci/controller/rpc.lua:53: attempt to index global 'luci' (a nil value)","code":-32602},"jsonrpc":"2.0"}

please reopen this ticket

SamJongenelen, 2015-12-15:
Same here, using `Package luci (git-15.248.30277-3836b45-1)`
and 
`Package luci-mod-rpc (git-15.248.30277-3836b45-1) `

SamJongenelen, 2015-12-15:
So,  I found the issue:
https://github.com/openwrt/luci/issues/576

Fix; set luci timeout in config, or set the timeout hard in the luci-mod-rpc line 55

jow-, 2015-12-15:
The fix in 16047d03c54df4474cb675a9ce98d06aad5a656b was incomplete and has been corrected with fff7c6c5188ff7a82b5d9f46a6f2c143288b5ed1.

jow-, 2015-12-15:
I also cherry-picked fff7c6c5188ff7a82b5d9f46a6f2c143288b5ed1 into the `for-15.05` branch now.