openwrt/luci #432: Showing MAC=00:00:00:00:00:00 at WAN&WAN6 on TL-WDR4900 (recent trunk + older versions)

qfqow, 2015-08-04:
Luci is showing at tab "interfaces" at WAN and WAN6 the mac address 00:00:00:00:00:00 (ifconfig show propper MAC). When i go into configuration and set the overwrite-mac to anything, then luci shows that. When i remove the overwrite-mac again, it shows again 00:00:00:00:00:00.
Bug is there in most recent trunk and it was there in trunk from before one month.

I checked on an other ralink device with 15.05-rc3 and the mac there was shown correctly.

Napsi-GER, 2015-10-05:
+1
Exactly the same for me. 15.05, r47065

Napsi-GER-

George-TL, 2016-03-27:
https://dev.openwrt.org/ticket/20606

cshoredaniel, 2017-01-14:
From https://dev.openwrt.org/ticket/20606

OpenWrt Designated Driver r49207 / LuCI Master (git-16.108.33174-51a7f96) and still experiencing this.

More details, I get zero wan mac on luci when wan protocol is pppoe, if I switch it to unmanaged, then I get the correct one.

Plus,ubus call network.device status also give me all zeros on wan mac when wan proto is set to pppoe.

This appears not to be a LuCI issue but core; can you confirm this happens with trunk of OpenWrt/LEDE?

LipkeGu, 2018-01-20:
I have also this situation on my fritzbox with openwrt (ifconfig) shows:

```
dsl0      Link encap:Ethernet  HWaddr 9C:C7:A6:E8:63:2B
          inet6 addr: fe80::9ec7:a6ff:fee8:632b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1504160 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1117396 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1449466463 (1.3 GiB)  TX bytes:101669023 (96.9 MiB)

dsl0.7    Link encap:Ethernet  HWaddr 9C:C7:A6:E8:63:2B
          inet6 addr: fe80::9ec7:a6ff:fee8:632b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1504160 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1117389 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1422391583 (1.3 GiB)  TX bytes:91622382 (87.3 MiB)

pppoe-wan Link encap:Point-to-Point Protocol
          inet addr:************  P-t-P:********  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1492  Metric:1
          RX packets:1289758 errors:0 dropped:0 overruns:0 frame:0
          TX packets:902970 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3
          RX bytes:1403064941 (1.3 GiB)  TX bytes:65323785 (62.2 MiB)

pptp-vpn  Link encap:Point-to-Point Protocol
          inet addr:10.20.0.12  P-t-P:10.20.0.102  Mask:255.255.255.255
          inet6 addr: fd04:2c74:3f98:3ab6:4015:6730:cfe4:70f/64 Scope:Global
          inet6 addr: fe80::4015:6730:cfe4:70f/10 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1400  Metric:1
          RX packets:20104 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1742 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3
          RX bytes:1461298 (1.3 MiB)  TX bytes:264346 (258.1 KiB)
```

It seems luci shows 00:00:00:00:00:00 due to the software generated interface ```pppoe-wan``` and has no access to the underlaying interface infoformation dsl0.*

SinglWolf, 2018-02-19:
Confirm the same problem.
Router: TP-LINK TD-W8970 v1.2
Firmware version: LEWA Reboot 17.01.4 r3560-79f57e422d / Lucille de-17.01 branch (git-17.290.79498-d3f0685)
Kernel version: 4.4.92

SinglWolf, 2018-09-20:
The problem still exists.
Router: TP-LINK TD-W8970 v1.2
Firmware version: OpenWrt SNAPSHOT r8080-5efd080e20 / LuCI Master (git-18.262.34590-1a0316b)
Kernel version: 4.14.70

feckert, 2022-12-16:
This This ticket is quite old and a lot has happened. I am therefore closing it. If the problem persists, then you have to look at it again.

closed 2022-12-16 by feckert