openwrt/luci #436: openvpn config tls_auth / key_direction

fkooman, 2015-08-10:
Hi,

Currently, the tls_auth is defined as this:

```
{ Value, "tls_auth", "/etc/openvpn/tlsauth.key 1", translate("Additional authentication over TLS") },
```

It should probably be a `FileUpload`. As for the `1` flag, this can also be set using `key_direction`. This allows an easy way to configure the `tls_auth` file without using SSH to do it.

Wedmer, 2015-08-11:
Ok, I'll change this to FileUpload. Also key_direction will be added.

Wedmer, 2015-08-11:
You should open ticket at https://dev.openwrt.org/ as openvpn package should be changed too.

fkooman, 2015-08-11:
Why does it need to be changed? Because it needs to be able to generate the tls-auth key you mean? Otherwise it can just point to the file uploaded to `/lib/uci/upload/` I guess?

Wedmer, 2015-08-11:
First of all key_direction option is not accepted by openvpn from core OpenWrt system, so I have to think about reimplementing some controller and view functionality.

fkooman, 2015-08-14:
I found I need one more option by experimenting and adding them myself directly to the lua code, in addition to `key_direction` also `tls_version_min` was needed. My current config:

```
config openvpn 'eduvpnnnn'
	option nobind '1'
	option tls_cipher 'TLS-DHE-RSA-WITH-AES-128-GCM-SHA256:TLS-DHE-RSA-WITH-AES-256-GCM-SHA384'
	option client '1'
	option comp_lzo 'yes'
	option dev 'tun'
	option remote_cert_tls 'server'
	option verb '3'
	option cipher 'AES-256-CBC'
	option persist_key '1'
	option auth 'SHA256'
	option tls_version_min '1.2'
	option key_direction '1'
	option persist_tun '1'
	option remote 'dev.eduvpn.org'
	option cert '/lib/uci/upload/cbid.openvpn.eduvpnnnn.cert'
	option key '/lib/uci/upload/cbid.openvpn.eduvpnnnn.key'
	option ca '/lib/uci/upload/cbid.openvpn.eduvpnnnn.ca'
	option tls_auth '/lib/uci/upload/cbid.openvpn.eduvpnnnn.tls_auth'
	option enabled '1'
```

The modifications to `/etc/init.d/openvpn` were easy, I just added `key_direction tls_version_min` to `append_params`, that worked. 

Changes in `openvpn-advanced.lua` under `Cryptography`:
```
{ Value,                "tls_version_min",				"1.0",  translate("minimum TLS version supported")},
{ FileUpload,		"tls_auth",						"/etc/openvpn/tlsauth.key",				translate("Additional authentication over TLS") },
```

I could send a pull request for this if you are interested, or maybe this needs more thinking. Let me know if I can help!

Wedmer, 2015-08-14:
It's easy to edit it in place, but `/etc/init.d/openvpn` locates at main OpenWrt repository. So you have to create a ticket to request changes. I'm not official  luci-openvpn maintainer. I'm just a person who made it to work with current version of OpenWrt. So you freely add your pull request but if it's merged it would break system for others.

fkooman, 2015-08-15:
I found it! I'll create a pull request there as well. As far as I know those two changes are independent of each other. If one modifies luci without modifying the init script it just won't work (like now), and if one modifies the init script without modifying luci one is not able to set it in the GUI, but only in the configuration file manually...

fkooman, 2015-08-28:
I sent an email to mirko@, as he is the maintainer of the OpenVPN package it seems, requesting the change. I am not subscribed to openwrt-devel@ so the mail there bounced :(

fkooman, 2015-08-28:
Okay, resent.

https://lists.openwrt.org/pipermail/openwrt-devel/2015-August/035456.html

tf0054, 2016-04-18:
Hi, the patch for init script hasnt been accepted yet, have it?

Of course we can modify the config file manually, but..
> option tls_auth '/lib/uci/upload/cbid.openvpn.eduvpnnnn.tls_auth 1'

0x90, 2016-08-22:
key_direction option seems not working. Does'not exist in generated config file /var/etc/openvpn-myvpn.conf

theftf, 2016-10-26:
Are there any news on this?

Wedmer, 2016-10-28:
> key_direction option seems not working. Does'not exist in generated config file /var/etc/openvpn-myvpn.conf

As main OpenWrt repositry moved to github you can create issue or make changes and create pull-request
https://github.com/openwrt/openwrt/tree/master/package/network/services/openvpn

dibdot, 2018-11-21:
Please re-test with latest version and open a new issue if something is still pending.

closed 2018-11-21 by dibdot