openwrt/luci #439: luci_statistics =>config wrong

miniminime, 2015-08-16:
Hi,

I think this is wrong: option 'Include' '/etc/collectd/conf.d'
I think it should read: option 'Include' '/etc/collectd.conf' 

I am also not able to add/change values. It always gets overwritten, but that might be a shortcoming on my side.

Cheers

mmunz, 2015-08-16:
No, this is Include is correct, additional collectd config files may be placed in /etc/collectd/conf.d/ directory. The reason why  /etc/collectd.conf is overwritten is that it is created by the /etc/init.d/luci_statistics init script. So either add your changes in /etc/config/luci_statistics or /etc/collectd/conf.d (untested by me, but i guess it works).

closed 2015-08-16 by mmunz

miniminime, 2015-08-18:
Cool, many thanks for your reply!