openwrt/luci #443: luci_statistics - dependency and module issues

miniminime, 2015-08-18:
Hi,

I think there is an issue with dependencies during installation. Further, if you try to install additional modules like mod-sensors or mod-thermal, it doesn't work/show-up in luci.

[image]

jow-, 2015-08-18:
The plugin load messages are normal and mod-thermal is not supported in the gui.

closed 2015-08-18 by jow-

miniminime, 2015-08-18:
Any chance that it could get supported?

itrankolov, 2015-08-21:
I have it working. Tell me where to send the modified files so you can add it?

miniminime, 2015-10-03:
@jow- you are a strange maintainer, really. A lot of packages totally outdated and even if someone is offering you the solution you just don't seem to care...but I know the default answer, I'm doing this in my free time and voluntarily, blablabla.

jow-, 2015-10-03:
@miniminime - I'd gladly accept PRs or patches implementing new plugins but I simply don't have the time to work on it, nor do I have a working test setup to try the thermal plugin. As for the question where to send the working solutions, that is already covered in this repos readme which links to https://github.com/openwrt/luci/blob/master/CONTRIBUTING.md

miniminime, 2015-10-03:
@jow- call me impressed and I'm sorry for my previous very unfair and provoking comment
@itrankolov could you provide your solution?