openwrt/luci #446: luci-app-ocserv not setting up Firewall

uk3, 2015-08-19:
Hello,

I used luci-app-ocserv to set up an openconnect server. Everything was set up correctly, except for the firewall. the Firewall Zone setting did not really do anything.

The only fix so far was to add the following to /etc/config/network:
`config interface 'VPN'
	option proto 'none'
	option ifname 'vpns+'
`

and to add the VPN interface to the lan zone.

Am I missing something here?

uk3, 2015-08-19:
The workaround is (unknown to me) documented here anyway: https://github.com/openwrt/packages/tree/master/net/ocserv

So it is just the luci app that does not do that as it seems.

btw, i am on trunk.

cshoredaniel, 2017-01-10:
No action for years; closing

closed 2017-01-10 by cshoredaniel