openwrt/luci #453: Should changing Network>Hostnames result in /etc/init.d/dnsmasq restart?

c3iq, 2015-08-27:
When I add or change a host's IP in Network>Hostnames I hope that when I hit Save & Apply that the change happens straight away.  I have to SSH in and run /etc/init.d/dnsmasq restart (or use a web script I've written).

I'm using:

LuCI Master (git-15.167.71950-8005ecb) / OpenWrt Chaos Calmer 15.05-rc3

Is my expectation wrong?  Is it a caching issue elsewhere?

cshoredaniel, 2017-01-14:
@c3iq Can you verify this happens with LuCI and OpenWrt/LEDE trunk; I'm not sure it's still an issue.  Also, can you look at /var/etc/dnmasq*.conf and verify (even on 15.05) that the file is or is not updated?  If is, then restart was done and the issue is that DNS may have been cached on your client.

jow-, 2018-06-25:
Procd reload triggers should take care of this nowadays.

closed 2018-06-25 by jow-