openwrt/luci #458: trim hostname in lucy input field

rubo77, 2015-08-30:
When entering a hostname starting or ending with a space, you get an error message, that there are invalid characters, but for a fast-paced person it is not really visible where.

This ist the same with many other input fields.

It would be handy if all field entries that don't allow the space or tap character would automatically be trimmed before validation on save so you only get an error message if you do something really wrong.

rubo77, 2015-12-03:
Can we add this?

Or which module would I have to modify, to achieve this myself?

jow-, 2015-12-03:
You need to edit `cbi.js` and `datatypes.lua`

rubo77, 2015-12-04:
related: https://github.com/freifunk-gluon/gluon/issues/414#issuecomment-161831581

jow-, 2018-07-03:
Personally, I am against auto-trimming inputs, especially due to the added complexity for fields that *do* accept spaces.

If someone proposes a clean and manageable change implementing this I might reconsider. As of now, I don't have plans to implement that.

closed 2018-07-03 by jow-