openwrt/luci #459: Luci front page auto-refresh is CPU heavy for the router

hnyman, 2015-08-31:
I have felt some sluggishness on the front page every now and then. I have also noticed some forum discussion on "slow Luci" ( https://forum.openwrt.org/viewtopic.php?id=59113 ).

I have also noticed that if the front page is left open on the browser with the default setting "auto refresh on", the CPU load in statistics grows surprisingly high. Even in my rather powerful WNDR3700, the CPU load goes over 20% just by having that front page open in browser...

Below is a graph highlighting this, tested with both Firefox and IE.
* Router is otherwise practically idle when I open the Luci front page and leave that page open. CPU load shoots to 20-25%, while the page updated dynamic content every 5 secs. 
* When moving to another page, the load immediately stops. 
* My front page has three XHR sections: main, DDND and UPNP. So I further tested uninstalling the Luci DDNS and MiniUPNP components, as their data is updated dynamically too. That cut the CPU load into half. So, the main Luci front page already causes rather high load by itself, but having three auto-refreshing sections really kills the CPU.

[image]

When profiling the front page with browser tools, the updating looks heavy. 
* When all three components are enabled, even the main data update takes almost 1.2 seconds (every 5 seconds). And there are 2 requests per cycle, main, DDNS and UPNP.
[image]
* If DDNS and UPNP components are disabled, the update duration decreases to 0.5s. So, having three updating components, slows things downs significantly
[image]

I have not yet tried to debug which section is the culprit, or if there is something too iterative in the lua script.

Jow, 
would it be ok to decrease the XHR update frequency from 5 secs to 10 secs?
Or any ideas about further debugging?

jow-, 2015-08-31:
Yes, I'm fine with increasing the timeout.

chris5560, 2015-08-31:
For luci-app-ddns I see no problem to increase the poll time.
If you find a good poll interval I'll implement it in next version.
I'm currently working on an extension of ddns-scripts/luci-app-ddns.

Christian

gwlim, 2015-09-14:
Can you compared it to BB?
I don't think Poll Interval is the root cause.

thesabbir, 2015-09-27:
A system/administration setting perhaps? Where we can specify the intervals.
As #483 wants to increase it.

hnyman, 2015-09-27:
The real-time graph poll interval mentioned in #483 is mostly a separate issue, as that 3 seconds  concerns only the update frequency of the four graphs in Status--> Realtime Graphs. If there would be a config option for the refresh intervals, there should probably be two settings: one for generic status pages, one for dedicated real-time info pages like that "realtime graphs".

And yes, the poll interval is not the root cause, but a small interval increases the visibility of the underlying problem: something in the polling mechanism causes high load.

jow-, 2018-06-25:
This was likely caused by inefficient status page plugins.

closed 2018-06-25 by jow-