openwrt/luci #469: tp-link wr1043nd v2

taoloo897, 2015-09-11:
Hi
I put your firmware on my router.
Please help me with nest issues:
1.i set a root password but the router on every page told me 
"No password set!
There is no password set on this router. Please configure a root password to protect the web interface and enable SSH.
Go to password configuration"

2.i can login with my new password and without password

3. i setted my pppoe user and pass.i connected to my provider, but after reboot the rooter these settings disapper , not saved

4.if i install a torrent client , it show in my software list , but after reboot it is deleted.

5.how can i put my factory firmware back, because if i upgrade from web ui interface , the router after reset begin with your firmware again

MANY MANY THANKS FOR YOUR HELP AND TIME

sh2d, 2015-10-22:
OpenWrt 15.05 Chaos Calmer runs fine here on a TP-Link TL-WR1043ND v2.  

1. What “firmware” did you put on your router (URL to .bin file)?
2. How did you “set” root password (detailed steps)?

hnyman, 2016-03-12:
Sounds like not a luci issue.

closed 2016-03-12 by hnyman