openwrt/luci #490: Comfortable WAN/internet access configuration

Jens14L, 2015-09-27:
I like the idea of free firmwares behind OpenWRT, but I would appreciate a comfortable menu (like the menu of the Telekom, cf. manual) to configure the login credentials and connect to the WAN/internet. This could expand the distribution and popularity of OpenWRT I think.

I have tested OpenWRT 12.09 on my Speedport W 502V.

* http://wiki.openwrt.org/toh/t-com/spw502v
* https://hilfe.telekom.de/dlp/eki/downloads/Speedport/Speedport_w502v_type_a/Bedienungsanleitung_Speedport_W502V.pdf#page=35

Jens14L, 2015-12-27:
Is there an easy way for DSL/Internet connections like in AVM Fritz!Boxes (username + password)?

jow-, 2015-12-28:
No, not really. Telekom routers are easy because they're tailored for T-DSL connections with most parameters already preconfigured and the firmware made specifically for that single model.

OpenWrt is a generic distribution targeting many different devices and wan access methods, therfore it requires more effort to setup properly.

cshoredaniel, 2016-12-10:
Not really a bug or specific enhancement; efforts are being made for next gen luci, but that will take quite some time.

closed 2016-12-10 by cshoredaniel