openwrt/luci #499: Issue while login (500 Internal Server Error)

Nappp, 2015-10-02:
I've been run into this problem for such a long time and I hasn't find a way to help and it is freaking me out so pls help me.

---
Env:
OpenWrt Attitude Adjustment (12.09 final)
[Feeds Using](https://gist.github.com/Nappp/fb7033423a255583dc31#file-gistfile1-txt)
Freshly build

---
Problem:
<b>500 Internal Server Error</b> when opens the login page, and I execute <code>/www/cgi-bin/luci</code> I captured the error as follows:
[Gist file](https://gist.github.com/Nappp/9b06c85dd01dd0c7623a#file-gistfile1-txt)

and when I cat out the <code>header.html</code> and read it, i found the issue maybe on this line:
```lua
local boardinfo = util.ubus("system", "board")
```

I have necessary <code>libubus</code> and <code>libuci</code> installed and I posted my opkg list here:
[opkg list file](https://gist.github.com/Nappp/fea6c289980f49684938#file-gistfile1-txt)

---

Is that anything to do with the retired 12.09 os?

hnyman, 2015-10-02:
Probably the main error is that you are using the feeds for Designated Driver trunk, instead of 12.09 versions.

Your feeds:
```
src-git packages https://github.com/openwrt/packages.git
src-git routing https://github.com/openwrt-routing/packages.git
src-git luci https://github.com/openwrt/luci.git
src-git telephony http://git.openwrt.org/feed/telephony.git
src-git management https://github.com/openwrt-management/packages.git
```
that should be
```
src-svn packages https://dev.openwrt.org/browser/branches/packages_12.09
src-git luci https://github.com/openwrt/luci.git;luci-0.11
src-git routing git://github.com/openwrt-routing/packages.git;for-12.09.x
```

hnyman, 2015-10-02:
https://dev.openwrt.org/browser/branches/attitude_adjustment/feeds.conf.default gives the Luci feed as
```
src-svn packages svn://svn.openwrt.org/openwrt/branches/packages_12.09
src-svn luci http://svn.luci.subsignal.org/luci/branches/luci-0.11/contrib/package
src-git routing git://github.com/openwrt-routing/packages.git;for-12.09.x
```
I am not sure if both work.

In general, AA12.09 is completely unsupported. The have been both BB14.07 and CC15.05 releases after it.

closed 2015-10-02 by hnyman

Nappp, 2015-10-04:
The first luci address works just fine, compiled and runs normal again, and btw the whole ```svn.luci.subsignal.org``` site seems to be down. Sorry about 12.09 but my router is as old as the Attitude Adjustment. Thanks a lot.

sannyi, 2016-12-03:
Same error here, I can't add any firewall options in here. Any idea?
`/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute function dispatcher target for entry '/admin/network/firewall'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute arcombine dispatcher target for entry '/admin/network/firewall/zones'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/tblsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/firewall_zoneforwards'.
A runtime error occured: /usr/lib/lua/luci/model/firewall.lua:83: bad argument #2 to 'get' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:168: in function