openwrt/luci #500: luci-app-qos not compatible with latest Chaos Calmer 15.05

shanicoolest, 2015-10-05:
luci-app-qos is not working on latest openwrt cc 15.05.
I had to --force install to get it working. 

Here is the log.
Log:
root@OpenWrt:/# opkg install --force-depends luci-app-qos
Installing luci-app-qos (git-15.248.30277-3836b45-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/lantiq/xway/packages/luci/luci-app-qos_git-15.248.30277-3836b45-1_all.ipk.
Multiple packages (kmod-ipt-core and kmod-ipt-core) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipt-core and kmod-ipt-core) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipt-conntrack and kmod-ipt-conntrack) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipt-core and kmod-ipt-core) providing same name marked HOLD or PREFER. Using latest.
Installing qos-scripts (1.2.1-7) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/lantiq/xway/packages/base/qos-scripts_1.2.1-7_all.ipk.
Installing tc (4.0.0-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/lantiq/xway/packages/base/tc_4.0.0-1_lantiq.ipk.
Installing kmod-sched-core (3.18.20-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/lantiq/xway/packages/base/kmod-sched-core_3.18.20-1_lantiq.ipk.
Installing kmod-sched-connmark (3.18.20-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/lantiq/xway/packages/base/kmod-sched-connmark_3.18.20-1_lantiq.ipk.
Installing kmod-ipt-conntrack-extra (3.18.20-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/lantiq/xway/packages/base/kmod-ipt-conntrack-extra_3.18.20-1_lantiq.ipk.
Installing kmod-ifb (3.18.20-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/lantiq/xway/packages/base/kmod-ifb_3.18.20-1_lantiq.ipk.
Installing iptables-mod-ipopt (1.4.21-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/lantiq/xway/packages/base/iptables-mod-ipopt_1.4.21-1_lantiq.ipk.
Installing kmod-ipt-ipopt (3.18.20-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/lantiq/xway/packages/base/kmod-ipt-ipopt_3.18.20-1_lantiq.ipk.
Installing iptables-mod-conntrack-extra (1.4.21-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/lantiq/xway/packages/base/iptables-mod-conntrack-extra_1.4.21-1_lantiq.ipk.
Configuring kmod-sched-core.
Configuring kmod-ipt-conntrack-extra.
failed to find a module named xt_connbytes
failed to find a module named xt_connlimit
failed to find a module named xt_connmark
failed to find a module named xt_helper
failed to find a module named xt_recent
Configuring kmod-sched-connmark.
Configuring tc.
Configuring kmod-ifb.
Configuring kmod-ipt-ipopt.
failed to find a module named xt_dscp
failed to find a module named xt_DSCP
failed to find a module named xt_length
failed to find a module named xt_statistic
failed to find a module named xt_tcpmss
failed to find a module named xt_CLASSIFY
failed to find a module named ipt_ECN
failed to find a module named xt_ecn
failed to find a module named xt_hl
failed to find a module named xt_HL
Configuring iptables-mod-ipopt.
Configuring iptables-mod-conntrack-extra.
Configuring qos-scripts.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
iptables: No chain/target/match by that name.
Configuring luci-app-qos.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-qos:
 *      kernel (= 3.18.20-1-21fc05610f2d9a062b249b93a28b2207) *         kernel (= 3.18.20-1-21fc05610f2d9a062b249b93a28b2207) *         kernel (= 3.18.20-1-21fc05610f2d9a062b249b93a28b2207) *         kernel (= 3.18.20-1-21fc05610f2d9a062b249b93a28b2207) *        kernel (= 3.18.20-1-21fc05610f2d9a062b249b93a28b2207) *
root@OpenWrt:/#

hnyman, 2015-10-05:
Those errors come from the qos-scripts package, not luci-app-qos. The luci app just install the dependencies (qos-scripts, tc, ...), as you did not yet have the actual qos installed.

Are you trying to install packages from the release repo to your own private build? With kernel modules that is rather difficult.

cshoredaniel, 2017-01-14:
No reply for over a year; closing.

closed 2017-01-14 by cshoredaniel