openwrt/luci #502: PXE

parasense, 2015-10-05:
Let's talk about enabling a "PXE" dialog under the Network area some place, probably DHCP/DNS options (aka dnsmasq settings). 

What we want to do is wrap around the '/etc/config/dhcp' configuration parameters for PXE boot.
Like so:

config boot
        option servername 'hostname-of-tftpd-goes here'
        option serveraddress 'ipv4-of-tftpd-goes-here'
        option filename 'pxelinux.o'

That configured OpenWRT to point at a TFTPD server, and it could be a remote system on the LAN, or even the OpenWRT router itself. There is a built-in tftpd service that goes along with dnsmasq, and it really orthogonal to these parameters. One should be able to have a tftpd on OpenWRT, but have PXE point elsewhere on the network should that be desired, or just point of the local dnsmaq built-in tftpd. The point being one could have a bazzillion tftpd services setup around their WAN, but here we configure where the DHCP PXE system points to for network booting. Obviously these parameters only make sense if luci is operating a dhcp service that can do PXE stuff.

LipkeGu, 2016-12-22:
It seems he means the bootmenue something like this?

[image]

LipkeGu, 2018-01-18:
I'm thinking about to put this again up when im get it right up again i can fill a pull request with this changes but this will spread in about 3 or more PRs.

jow-, 2020-02-23:
Closing this as its not supported by OpenWrt and never gained traction. PRs welcome.

closed 2020-02-23 by jow-

vKnmnn, 2020-03-22:
There is one feature request for this feature. See #3356 
I would also like to see this feature in OpenWRT.