openwrt/luci #506: luci-lib-json: module 'ubus' not found

corny, 2015-10-09:
I built a chaos calmer with `luci-lib-json` and I am unable to require `luci.json`:

```lua
> local json = require('luci.json')
/usr/lib/lua/luci/util.lua:13: module 'ubus' not found:
	no field package.preload['ubus']
	no file './ubus.lua'
	no file '/usr/share/lua/ubus.lua'
	no file '/usr/share/lua/ubus/init.lua'
	no file '/usr/lib/lua/ubus.lua'
	no file '/usr/lib/lua/ubus/init.lua'
	no file './ubus.so'
	no file '/usr/lib/lua/ubus.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/util.lua:13: in main chunk
	[C]: in function 'require'
	/usr/lib/lua/luci/json.lua:6: in main chunk
	[C]: in function 'require'
	stdin:1: in main chunk
	[C]: ?
```

```
$ opkg list-installed | grep luci
luci-base - git-15.276.51144-8832d53-1
luci-lib-ip - git-15.276.51144-8832d53-1
luci-lib-json - git-15.276.51144-8832d53-1
luci-lib-nixio - git-15.276.51144-8832d53-1
```

Are any dependencies missing?

corny, 2015-10-09:
The dependency `libubus-lua` should be added to the Makefile.

cshoredaniel, 2016-12-10:
Appears to be an old issue.  Builds work on all current releases and trunk, so need details on what is failing.

cshoredaniel, 2016-12-10:
Can you verify if this occurs with current trunk?  Also, does it happen with current CC?

commit 8ca29efe (2018-07-03): luci-lib-json: depend on luci-base

Fixes #506
closed 2018-07-03 by jow-