openwrt/luci #511: luci on lighttpd not working

DocMAX, 2015-10-10:
i followed the instrucions on

​http://wiki.openwrt.org/doc/howto/luci.on.lighttpd

but it doesnt work:

/usr/lib/lua/luci/dispatcher.lua:526: attempt to index local 'f' (a nil value)
stack traceback:
/usr/lib/lua/luci/dispatcher.lua:526: in function 'createindex'
/usr/lib/lua/luci/dispatcher.lua:534: in function 'createtree'
/usr/lib/lua/luci/dispatcher.lua:202: in function 'dispatch'
/usr/lib/lua/luci/dispatcher.lua:169: in function </usr/lib/lua/luci/dispatcher.lua:168>

jow-, 2015-10-12:
Make sure that Lighttpd is running as root, otherwise LuCI will nto function properly.

cshoredaniel, 2016-12-10:
See #734 .  Marking as duplicate as #734 has more information.

closed 2016-12-10 by cshoredaniel