openwrt/luci #514: luci on nginx

Sergi0, 2015-10-18:
Is it possible to make luci work on nginx?
luci.on.nginx page is missing on OpenWRT website, and a few configs I have found don't work (OpenWRT 14.07) and return 404 Not Found for 192.168.1.1/cgi-bin/luci

Configs I have tried:
        location /cgi-bin/luci {
            fastcgi_split_path_info ^(/cgi-bin/luci)(/.+)$;
            fastcgi_pass  unix:/var/run/luafcgid.sock;
            fastcgi_param PATH_INFO $fastcgi_path_info;
            fastcgi_param SCRIPT_FILENAME /usr/lib/lua/luci/sgi/fcgi.lua;
            include       fastcgi_params;
            client_max_body_size 16M;
        }

         location /cgi-bin/luci {                                             
                fastcgi_split_path_info ^(/cgi-bin/luci)(/.+)$;                  
                fastcgi_pass  unix:/tmp/fcgi-cgi.sock;                           
                fastcgi_param PATH_INFO $fastcgi_path_info;                      
                set $script_name '/cgi-bin/luci';                                
                fastcgi_param SCRIPT_FILENAME $document_root$script_name;
                include fastcgi_params;                                    
                add_header Cache-Control no-store;
        }

275761919, 2016-08-08:
do you solve this problem? i also want to run luci on nginx

hongkongkiwi, 2016-10-03:
Also keen on this, having the same problem.

cshoredaniel, 2016-12-10:
I'd suggest you prefer using uhttpd on a different port than nginx and keep the admin UI separate from you nginx instance.  There is no advantage to nginx as your UI server.

Pietia10, 2016-12-29:
I think we really need to set luci as default for serving luci as uhttpd take like 5-10 seconds to load any page(100% CPU in top for just one user!),
I think first we need to solve this
https://github.com/openwrt/packages/issues/2408
so nginx can serve old plain cgi scripts.

zhanhb, 2018-05-13:
Any progress about this?

closed 2018-05-13 by jow-