openwrt/luci #516: Token removal from URL prevents login with IE11

hnyman, 2015-10-21:
After the change in https://github.com/openwrt/luci/commit/86326e0deff92a485ffd47e22ac70194abb3fd66 I can login ok with Firefox, but not with IE11. The browser remains on the login screen.

If a page address is entered manually, it works and also navigation afterwards.
e.g. https://192.168.1.1/cgi-bin/luci/admin/system/startup

So that sounds like a browser-specific problem on the login page redirecting functionality.

Schabram, 2015-10-21:
Have the same problem with chrome but not after a reset change few thinks with LUCI and then locked out.

First login needed to trys.

jow-, 2015-10-21:
Did you clear your browser caches to rule out redirects to stale urls and the like?

Schabram, 2015-10-21:
yes cleared its clear, i test chrome last stable version with Windows and Linux google-chrome

hnyman, 2015-10-21:
I agree. I applied manually b4d62d3fd7c7093828c84b5f408cfb0ff6d84939 and https://github.com/openwrt/luci/commit/79383f5a74bcff528a5270a930c36b58e310010c and login with IE11 works now ok.

hnyman, 2015-10-21:
Hmmm. I am reusing this issue to report bad news:

I compiled a new version and tried to flash it, but flash window timeouts in the image upload & checksum calculation phase (after selecting firmware file and pressing "Flash image" button) with this error:
   Bad Gateway
   The process did not produce any response

Luci should be up-to-date, as I manually applied the two commits after the previous build.

It timeouts both in Firefox and in IE.

thess, 2015-10-21:
I concur with this - flash upload hangs since last update. However, I can login OK now.

hnyman, 2015-10-22:
After 94ab57f48c9a8704a0a7da87bfa6ced9e73ca9be flashing works again.
I close this ticket (again), but will leave #517 open for @kbabioch to verify and close.

closed 2015-10-22 by hnyman