openwrt/luci #517: Firmware cannot be upgrade via web interface any longer

kbabioch, 2015-10-21:
On my WNDR3700v2 (ar71xx, 16MB flash) I am no longer able to update the firmware using the LUCI web interface. I only get a "Bad gateway" response, whereas before I would get to the overview page showing me the size & hash and asking me to proceed. As far as I can tell a recent change (something after Oct 13) must have broken this, so it should be not too much of a hassle to find out what the issue is.

kbabioch, 2015-10-21:
Every now and then I'm also getting the following trace rather than a "Bad gateway" response:

    not enough memory
    stack traceback:
	/usr/lib/lua/luci/http/protocol.lua:123: in function '__appendval'
	/usr/lib/lua/luci/http/protocol.lua:320: in function 'store'
	/usr/lib/lua/luci/http/protocol.lua:397: in function 'snk'
	/usr/lib/lua/luci/ltn12.lua:300: in function 'step'
	/usr/lib/lua/luci/ltn12.lua:309: in function </usr/lib/lua/luci/ltn12.lua:305>
	(tail call): ?
	(tail call): ?
	/usr/lib/lua/luci/http.lua:95: in function '_parse_input'
	/usr/lib/lua/luci/http.lua:39: in function </usr/lib/lua/luci/http.lua:37>
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:183: in function 'require_post_security'
	/usr/lib/lua/luci/dispatcher.lua:410: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:169: in function </usr/lib/lua/luci/dispatcher.lua:168>

The `not enough memory` message is worrysome, since I should have something around 40 MB of free RAM and ~12 MB of free flash.

kbabioch, 2015-10-21:
Bisecting through the recent changes, I've found the culprit: b5826f1ffb19288b8bfdc63f4b77700cfabe1181

The commit message clearly states that you've changed things around and wanted to make it resistent against CSRF. Unfortunately I'm not too familiar with Lua, so I will have to take a deeper look.

jow-, 2015-10-22:
I'm working on it, needs a further bit of restructuring.

hnyman, 2015-10-27:
I close the issue as to my kwnoledge the problem has been solved with the commit 94ab57f48c9a8704a0a7da87bfa6ced9e73ca9be where Jow referenced this issue. I have since then updated my own WNDR3700 several times in Luci.

But if you currently have the problematic Luci version, you need to either 
* download a new firmware to the router (e.g. with wget) and use "sysupgrade" to flash it, or 
* download the fix commit to the router as a patch and use "patch" to apply it to Luci.
https://github.com/openwrt/luci/commit/94ab57f48c9a8704a0a7da87bfa6ced9e73ca9be.patch

closed 2015-10-27 by hnyman