openwrt/luci #519: [Off Topic] gLuCI and qLuCI

GreenLunar, 2015-10-22:
To be named, per binary, as luci-gtk and luci-qt.

With an ambition of releasing excessive load on router devices due to LuCI web interface, did you or any other OpenWrt project, such as Gargoyle and Tomato, think of writing an end-user-friendly desktop client for the OpenWrt platform?

jow-, 2015-10-22:
I did and deemed it too much work.

closed 2015-10-22 by jow-

GreenLunar, 2015-10-22:
I suggest to open a discussion in this concern.  Such initiative might embodied faster than you might think.

For too long, OpenWrt is dependent on Lua by LuCI, as it is.

jow-, 2015-10-22:
It would make more sense to open this discussion on the OpenWrt mailing lists or forums, not on the LuCI issue tracker.

GreenLunar, 2015-10-22:
I will do so.

Be ready to initiate https://github.com/OpenWrt/gLuCI and https://github.com/OpenWrt/qLuCI