openwrt/luci #520: Theme Material for lat update LuCI* is not working properly

build000, 2015-10-22:
http://i.imgur.com/DDq7XrL.png

hnyman, 2015-10-22:
Yes, the security token handling changes have impacted negatively on Material theme.

I have already filed a ticket for the developer at his Github site:
https://github.com/LuttyYang/luci-theme-material/issues/18

hnyman, 2015-10-22:
The issue is apparently using the existence of ";" in the URL to sniff if the user is logged in or still at the login screen. In the new scheme there is no ";" in the URL, so Material gets it wrong.
https://github.com/LuttyYang/luci-theme-material/blob/master/files/htdocs/js/script.js#L147

I made a small patch in a live router to /www/luci-static/material/js/script.js . I eliminated the sniffing for ";". This works for me otherwise ok, but there is an empty left menu visible at the login screen:

```
root@OpenWrt:/usr/lib/lua/luci# diff /rom/www/luci-static/material/js/script.js
/www/luci-static/material/js/script.js
--- /rom/www/luci-static/material/js/script.js
+++ /www/luci-static/material/js/script.js
@@ -145,9 +145,6 @@
         var ret = false;
         var getUrlNode = function (href){
             var linkPos = href.indexOf(";");
-            if (linkPos == -1){
-                return "login";
-            }else{
                 linkPos = href.indexOf("/", linkPos);
                 if (linkPos == -1){
                     return "overview";
@@ -158,7 +155,6 @@
                     else
                         return link;
                 }
-            }
         };

         var currentNode = getUrlNode(window.location.pathname);

```

jow-, 2015-10-22:
@hnyman - thanks for digging into the code, that gave me enough clues to fix it.

closed 2015-10-22 by jow-