openwrt/luci #523: luci-app-diag-devinfo cannot be installed on 15.05 Final on Linksys WRT1900AC

sorinello, 2015-10-26:
I am running CC 15.05 final on a mvebu device (Linksys WRT1900AC V1), with
 the correct opkg.conf

 I try to install luci-app-diag-devinfo but I get the following error:

```
root@xxx:/etc# opkg install luci-app-diag-devinfo
 Installing luci-app-diag-devinfo (git-15.248.30277-3836b45-1) to root...
 Downloading
 http://downloads.openwrt.org/chaos_calmer/15.05/mvebu/generic/packages/luci
 /luci-app-diag-devinfo_git-15.248.30277-3836b45-1_all.ipk.
 Collected errors:
  * satisfy_dependencies_for: Cannot satisfy the following dependencies for
 luci-app-diag-devinfo:
  *      smap *  netdiscover *   mac-to-devinfo *        httping *
 smap-to-devinfo *       netdiscover-to-devinfo *
  * opkg_install_cmd: Cannot install package luci-app-diag-devinfo.
```

 I tried installing each package manually, but it seems that NONE of them
 exists.

hnyman, 2015-10-27:
There is not much to fix on the Luci side, except possibly marking the app BROKEN and thus disabling it.

The core reason for the depenency error is that the dependency packages have never been imported into the new packages feed on Github, but are still in the "oldpackages" feed (the old site). That oldpackages feed is not built any more.

* All three "-to-devinfo" packages are provided by this:
  https://dev.openwrt.org/browser/packages/net/mac-to-devinfo/Makefile

* httping is also there:
  https://dev.openwrt.org/browser/packages/net/httping/Makefile

* Smap has been first moved to telephony and then declared abandoned:
  https://github.com/openwrt/packages-abandoned/blob/master/net/smap/Makefile

ok in trunk, but not in 15.05:
* Netdiscover
  https://github.com/openwrt/packages/tree/master/net/netdiscover

If you want those packages activated, you should assume maintainership for them and create "pull request" at the packages repo to import the packages: https://github.com/openwrt/packages/pulls

Read also https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md

If you just want the packages for yourself, you could checkout an Openwrt build environment, enable oldpackages feed there and build the packages for yourself.

hnyman, 2015-10-27:
I marked the app BROKEN at both Luci trunk and for-15.05 branch.

The app can be enabled again when all the dependency packages have been imported to packages repo.

closed 2015-10-27 by hnyman

dfc-d, 2015-12-07:
Actually it's more complicated than that.  The whole stack was never all that great to begin with (not my best work), and may not even worth any more even with the dependencies added.  At the very least for the part based on netdiscover there is Issue #234 which is that due to uhttpd script timeout of 60 seconds, the scan is likley to timeout with an error message rather than produce useful results.   That part would have to be rewritten to detach the scan from the cgi process so that that timeout wouldn't kill it, in addition the the need for a mechanism to just ajax to get the results as they accumulated, rather than current run a script and grab the results approach that fails because the results take longer than 60 seconds to be completed.