openwrt/luci #530: cbi Value ignore Property .size

chris5560, 2015-11-01:
Whatever I set for Value.size property, it's included in the final html code displayed by the browser but nothing changes in the display.
I checked with Firefox and OpenWrt, Bootstrap and Material themes.

cshoredaniel, 2016-12-10:
@chris5560 Is is the case with LuCI trunk, or was this on older versions?  IIRC the size parameter is for allowed number of character and the display size is a different setting (or done by CSS). @jow- am I remember that correctly?

chris5560, 2016-12-10:
@cshoredaniel @jow- I just checked on my current lede build r2438-a245772 with luci-app-ddns and openwrt, bootstrap and material theme. Size value is changed if you look into source of the called page but the display width doesn't change. I think this is somewhere hard coded inside stylesheets.
I'm not familiarly with css but if not set there should be a default otherwise use the given parameter.

I looked at http://www.w3schools.com/tags/att_input_size.asp
size == visible width, in characters
maxlength == maximum number of characters allowed

cshoredaniel, 2016-12-13:
@chris5560 Thanks for looking into this; I think I know enough CSS I think, and will dig into this, unless you think you'll have time (I have a far number of LuCI things I've assigned myself).

cshoredaniel, 2016-12-30:
@chris5560 I've confirmed it's the CSS that's the issue.  Got a fix for bootstrap and openwrt themes but not material at the moment.

cshoredaniel, 2016-12-30:
@chris5560 material specifies some minimum widths but no maximums.  How small are you trying to go?

chris5560, 2016-12-30:
@cshoredaniel
I have no real idea what is a good minimum value.
Inside luci-app-ddns I create my own detail_value.htm with style="width: 10em;" but it's ignored by "material".
I don't know what are the current defaults but I think 10 char minimum width should be ok.

cshoredaniel, 2017-01-14:
@chris5560 I haven't had a chance to figure out material yet, but have got boostrap, openwrt, and looked at freifunk (I don't think a change was needed there IIRC).

chris5560, 2017-01-15:
@cshoredaniel
thanks for feedback.
Currently I think a fix is not needed for LEDE-17.01 release.
I'm currently working on the next major release of ddns-scripts which require a rewritten luci-app-ddns, but this will not happen before LEDE-17.01 release.

jow-, 2018-07-03:
I consider the `.size` attribute deprecated - field sizes should be controlled via CSS.

closed 2018-07-03 by jow-