openwrt/luci #533: Logical error in recognising and displaying >40 MHz channel widths

hnyman, 2015-11-02:
Oirignal report  https://dev.openwrt.org/ticket/20840:
```
when the channel width for the interface is set to either 80Mhz or 160Mhz
 --all clients are shown as connecting on 20Mhz. 
```

Luci seems to have a logic error in the channel width recognition and display. The logic has likely been crafted when only 20 or 40 MHz were available. The erroneous logic goes this way: 
"if 40MHz flag is set, width is 40, otherwise it is 20"

Faulty logic is at least in:
admin_network/wifi_overview.htm
https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/luasrc/view/admin_network/wifi_overview.htm#L326-L333
```
assoclist[j].rx_40mhz ? 40 : 20
```

The flag is set (and alos used) in admin_status/index.htm:
https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm#L408
```
assoclist.push({
...
    rx_40mhz: net.assoclist[bssid].rx_40mhz,
```
https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm#L479-L486
```
assoclist[i].rx_40mhz ? 40 : 20
```

The error is in Luci trunk and earlier branches including for-15.05.

jow-, 2015-11-17:
That 20 || 40 formatting is the least problem here. In order to fix this, one needs to add VHT rate support to libiwinfo, then libiwinfo-lua first. Since I do not have any AC hardware available I can write a patch but someone else needs to test it.

jow-, 2016-01-25:
Fixed with these commits:

iwinfo
- http://git.openwrt.org/?p=project/iwinfo.git;a=commitdiff;h=1b471aa33dcc1031a456c3a04f04a0097e5b89a5
- http://git.openwrt.org/?p=project/iwinfo.git;a=commitdiff;h=e3bd6c3653669bd53b87944ecedebe5bce162f7a
- http://git.openwrt.org/?p=project/iwinfo.git;a=commitdiff;h=bc8682aa5ee323f5a6e272a28ef348f2f7534d55
- http://git.openwrt.org/?p=project/iwinfo.git;a=commitdiff;h=78f0fafa591754c3c5bbd3f408a0a91b78690141
- http://git.openwrt.org/?p=project/iwinfo.git;a=commitdiff;h=e4aca3910dff532ed878d0ceaf1ab6e8ad7719bf

OpenWrt
- https://dev.openwrt.org/changeset/48487
- https://dev.openwrt.org/changeset/48488

LuCI
- https://github.com/openwrt/luci/commit/4f47a18bb1a98ed263559e1cb40b5b5efc43f14b

closed 2016-01-25 by jow-