openwrt/luci #536: tiny improvement: make themes choosable instead of always forcing luci-theme-bootstrap

qjdor3, 2015-11-04:
Could you please make the themes choosable? At the moment luci force the users to always compile luci-theme-bootstrap with make menuconfig. It would be better if users can choose the theme by themself without beeing forced to always compile luci-theme-bootstrap.

This config for example is working without any problems:
# CONFIG_PACKAGE_luci-theme-bootstrap is not set
# CONFIG_PACKAGE_luci-theme-freifunk-generic is not set
CONFIG_PACKAGE_luci-theme-material=y
# CONFIG_PACKAGE_luci-theme-openwrt is not set

Why did i have to create such a config with vi by modifying .config? Does not make much sense.

jow-, 2015-11-04:
The theme is selected by the "luci" meta package which groups common components but does not include any functionality by itself. You can simply deselect "luci" and pick the desired components manually, this way you're not forced to include the bootstrap theme. The same applies to other dependencies like luci-app-firewall or luci-proto-ppp. Its simply not feasible to create metapackages for every possible combination of components.

closed 2015-11-04 by jow-

qjdor3, 2015-11-04:
Its not possible to include into the luci meta package something like a rule "at least there have to be choosen one theme" instead of forcing luci-theme-bootstrap?

jow-, 2015-11-04:
Unfortunately not as the kconfig language does not allow such expressions.

hnyman, 2015-11-04:
If you are doing just a personal build for yourself, you could easily edit the definition of "luci" itself in /collections:
https://github.com/openwrt/luci/blob/master/collections/luci/Makefile#L14
Just change +luci-theme-bootstrap to +luci-theme-material there, and "luci" contains that theme.

	+uhttpd +uhttpd-mod-ubus +luci-mod-admin-full +luci-theme-bootstrap \
	+luci-app-firewall +luci-proto-ppp +libiwinfo-lua +IPV6:luci-proto-ipv6

Other alternative might be to follow Jow advice about skipping "luci" metapackage, and simple selecting the eight components directly. (I am using that approach for the "luci-ssl", to switch SL to openssl and px5g to px5g-standalone.)

qjdor3, 2015-11-04:
Thanks jow-
Thanks hnyman. Thats what i would do now.