openwrt/luci #537: model.uci.Cursor:set_list(cfg,sect,opt,list) returns true when it doesn't do anything

karlp, 2015-11-05:
on both BB & CC:

luci.model.uci.Cursor:set_list returns true even when it doesn't succeed.
If the config file doesn't exist, set_list returns false, but if the file exists, but the section doesn't, it returns true without doing anything.  Likewise with commit and save.

```
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio (double int32)
> uci = require("luci.model.uci")
> ucic = uci.cursor()
> bsl = ucic:set_list("blah", "some_section", "some_option", { 1,2,3})
> =bsl
false
> bsl2 = ucic:set_list("remake", "some_section", "some_option", {1,2,3})
> =bsl2
true
> =ucic:save("remake")
true
> =ucic:commit("remake")
true
> 
```
Config files themselves:
```
root@eg-02D455:~# uci show remake  # blank file, but exists
root@eg-02D455:~# uci show blah    # file doesn't exist.
uci: Entry not found
root@eg-02D455:~#
```

cshoredaniel, 2017-01-14:
@karlp Is this an issue with trunk OpenWrt/LEDE and trunk LuCI?  What about 15.05? CC & BB are no longer supported.

karlp, 2017-01-16:
This is easy to test for anyone with a current system, I included all the code necessary for testing on the console, nothing special required.  I filed this in November 2015.  I'll check again on a current lede, but I don't have one of them up right now.

cshoredaniel, 2017-01-16:
@karlp I've looked at the luci.model.uci; this part is just a passthrough to to libuci-lua from what I saw, so any issues belong there (which is in core).

cshoredaniel, 2017-01-16:
That is luci.model.uci.cursor() is just uci.cursor() (from libuci-lua) and that is what set_list comes from.

cshoredaniel, 2017-01-16:
I'll double-check to be sure but at this point it looks like a core bug.

cshoredaniel, 2017-01-16:
Ok, that looks, like a LuCI addon, after all

karlp, 2018-08-08:
This is still a problem on 18.06 era luci.

```
root@eg-037BCC:~# uci show blah
uci: Entry not found
root@eg-037BCC:~# uci show someconfig
root@eg-037BCC:~# lua
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio (double int32)
> uci = require("luci.model.uci")
> ucic = uci.cursor()
> = ucic:set_list("blah", "some_section", "some_option", { 1,2,3})
false	Entry not found
> =ucic:set_list("someconfig", "some_section", "some_option", { 1,2,3})
true	nil    <<< succeeds
> =ucic:save("someconfig")
true   <<< succeeds
> =uci:commit("someconfig")
true	nil
> 
> 
root@eg-037BCC:~# uci show someconfig    <<<< didn't actually succeed
root@eg-037BCC:~#
```

uci - 2018-03-24-5d2bf09e-1
luci-base - git-18.212.55608-5ec7261-1

cshoredaniel, 2018-08-08:
> This is still a problem on 18.06 era luci.

Are you using the right state directory (-P) ?  LuCI doesn't use the
same one as uci cli by default.

Regards,

Daniel

karlp, 2018-08-08:
@jow- this might actually be a different one, I just decided to retest my old tickets while I was in bug mode :)  I'll retest it after the rpcd/libuci fixes regardless.

jow-, 2018-08-08:
@karlp - will be fixed by the same rpcd updates though:

```
root@jj:~# lua -lluci.model.uci -e 'print(luci.model.uci:set_list("notexists", "notexist", "test", { 1, 2, 3 }))'
false	Entry not found
root@jj:~# lua -lluci.model.uci -e 'print(luci.model.uci:set_list("network", "notexist", "test", { 1, 2, 3 }))'
false	Entry not found
root@jj:~# lua -lluci.model.uci -e 'print(luci.model.uci:set_list("network", "lan", "test", { 1, 2, 3 }))'
true	nil
root@jj:~# 

```

karlp, 2018-08-20:
confirmed this is fixed in 1806 era and later after the rpcd fixes.

closed 2018-08-20 by karlp