openwrt/luci #538: luci.model.uci.Cursor:set(cfg, sect, opt_or_sectype, nil) doesn't match docs

karlp, 2015-11-05:
http://luci.subsignal.org/api/luci/modules/luci.model.uci.html#Cursor.set says that you can create a section by providing nil to the fourth argument, and providing the section type as the third argument.  This fails however.

```
>  ucic = require("luci.model.uci").cursor()
> ucic:set("remake", "plugins", "plugins", nil) -- make section
stdin:1: bad argument #4 to 'set' (string expected, got nil)
stack traceback:
	[C]: in function 'set'
	stdin:1: in main chunk
	[C]: ?
> 
```

This may just be a documentation issue, I'm just looking for a way to ensure that a section exists if it doesn't already, and was trying this out.

chris5560, 2015-11-15:
To check if section exist and create it, here some lines from global.lua of luci-app-ddns
``` lua
local m = Map("ddns")

local ns = m:section( NamedSection, "global", "ddns")

-- section might not exist
function ns.cfgvalue(self, section)
	if not self.map:get(section) then
		self.map:set(section, nil, self.sectiontype)
	end
	return self.map:get(section)
end
```
hope this help

karlp, 2015-11-16:
@chris5560 that's only for CBI, which isn't relevant here, nor does it address the documentation issue that this is filed for.

jow-, 2015-11-16:
Yes, the function checks the number of arguments, not whether arg 4 is a string or nil. So calling with three args will yield the desired behaviour while a call with 4 args requires arg 4 to be a string.

karlp, 2015-11-16:
Right, but the docs say, for argument 4, "value: UCI value or nil if you want to create a section"

Maybe the docs could show the three arg and four arg form separately? or if that's hard, the function should check the arg instead of the count?

jow-, 2015-11-16:
I am aware of that, was just clarifying the situation.

closed 2015-11-16 by jow-

catiamatos1991, 2017-06-05:
How to do a uci set for config "wireless" section "radio0" option "disabled" value "0" ?

I'm trying to get a new ubus session and the do a uci set and a commit and i got ubus code [0] mas in reality nothing is done to my device cause my set operation command is not working. Anyone can help me?

curl -d '{ "jsonrpc": "2.0", "id": 1, "method": "call", "params": [ "00000000000000000000000000000000", "session", "login", { "username": "root", "password": secret } ] }'  -g link
curl -s -d '{"jsonrpc":"2.0","id":1,"method":"call","params":
["'$UBUS_SESSION'","uci","set",{"config":"'teste_wireless'","section":"'radio0'",
"values": { "disabled":1}}]}' -g link
curl -s -d '{"jsonrpc":"2.0","id":1,"method":"call","params":
["'$UBUS_SESSION'","uci","commit",{"config":"'teste_wireless'"}]}' -g 
link