openwrt/luci #540: Show more useful information in the "Associated Stations" table

mauermann, 2015-11-05:
It would be nice if the "Associated Stations" table would show more useful information. e.g. with dual band radios getting more common it would be interesting to see the radio a station is associated. Also, instead of remembering MAC addresses, I think it's better to see its hostname or IP address.

Attached screenshot shows my proposed changes with columns for hostname and radio added. Unfortunately because of table width I had to drop the Noise column. However, this is less important for me.

I'm sure it needs some work and discussion, so it's probably too early to submit a pull request... ;-)

[image]

mauermann, 2015-11-05:
Here is my diff for the screenshot above. It's currently only for status/overview, but should be possible for network/wireless, too.

```diff
--- view/admin_status/index.htm.bak	2015-11-05 09:52:41.000000000 +0100
+++ view/admin_status/index.htm	2015-11-05 13:35:07.000000000 +0100
@@ -403,6 +403,18 @@
 
 						for (var bssid in net.assoclist)
 						{
+							var macaddr = bssid.toLowerCase();
+							var host = '?';
+							<% if has_dhcp then %>
+							for (var lidx = 0; lidx < info.leases.length; lidx++)
+							{
+								if (info.leases[lidx].macaddr == macaddr) {
+									host = info.leases[lidx].hostname ? info.leases[lidx].hostname : info.leases[lidx].ipaddr;
+									break;
+								}
+							}
+							<% end %>
+							
 							assoclist.push({
 								bssid:    bssid,
 								signal:   net.assoclist[bssid].signal,
@@ -414,7 +426,9 @@
 								tx_mcs:   net.assoclist[bssid].tx_mcs,
 								tx_40mhz: net.assoclist[bssid].tx_40mhz,
 								link:     net.link,
-								name:     net.name
+								name:     net.name,
+								device:   dev.device,
+								host:     host
 							});
 						}
 					}
@@ -464,23 +478,24 @@
 					);
 
 					tr.insertCell(-1).innerHTML = assoclist[i].bssid;
+					tr.insertCell(-1).innerHTML = assoclist[i].host.length > 18 ? assoclist[i].host.substring(0,15)+"..." : assoclist[i].host;
 
 					tr.insertCell(-1).innerHTML = String.format(
 						'<a href="%s">%s</a>',
 							assoclist[i].link,
 							'%h'.format(assoclist[i].name).nobr()
 					);
+					tr.insertCell(-1).innerHTML = assoclist[i].device;
 
 					tr.insertCell(-1).innerHTML = String.format('%d <%:dBm%>', assoclist[i].signal).nobr();
-					tr.insertCell(-1).innerHTML = String.format('%d <%:dBm%>', assoclist[i].noise).nobr();
 
 					tr.insertCell(-1).innerHTML = (assoclist[i].rx_mcs > -1)
-						? String.format('%.1f <%:Mbit/s%>, MCS %d, %d<%:MHz%>', assoclist[i].rx_rate / 1000, assoclist[i].rx_mcs, assoclist[i].rx_40mhz ? 40 : 20).nobr()
+						? String.format('%.1f <%:Mbit/s%>, MCS %d, %d<%:MHz%>', assoclist[i].rx_rate / 1000, assoclist[i].rx_mcs, assoclist[i].rx_40mhz ? 40 : 20)
 						: String.format('%.1f <%:Mbit/s%>', assoclist[i].rx_rate / 1000).nobr()
 					;
 
 					tr.insertCell(-1).innerHTML = (assoclist[i].tx_mcs > -1)
-						? String.format('%.1f <%:Mbit/s%>, MCS %d, %d<%:MHz%>', assoclist[i].tx_rate / 1000, assoclist[i].tx_mcs, assoclist[i].tx_40mhz ? 40 : 20).nobr()
+						? String.format('%.1f <%:Mbit/s%>, MCS %d, %d<%:MHz%>', assoclist[i].tx_rate / 1000, assoclist[i].tx_mcs, assoclist[i].tx_40mhz ? 40 : 20)
 						: String.format('%.1f <%:Mbit/s%>', assoclist[i].tx_rate / 1000).nobr()
 					;
 				}
@@ -676,14 +691,15 @@
 		<tr class="cbi-section-table-titles">
 			<th class="cbi-section-table-cell">&#160;</th>
 			<th class="cbi-section-table-cell"><%:MAC-Address%></th>
+			<th class="cbi-section-table-cell"><%:Host%></th>
 			<th class="cbi-section-table-cell"><%:Network%></th>
+			<th class="cbi-section-table-cell"><%:Radio%></th>
 			<th class="cbi-section-table-cell"><%:Signal%></th>
-			<th class="cbi-section-table-cell"><%:Noise%></th>
 			<th class="cbi-section-table-cell"><%:RX Rate%></th>
 			<th class="cbi-section-table-cell"><%:TX Rate%></th>
 		</tr>
 		<tr class="cbi-section-table-row">
-			<td colspan="7"><em><br /><%:Collecting data...%></em></td>
+			<td colspan="8"><em><br /><%:Collecting data...%></em></td>
 		</tr>
 	</table>
 </fieldset>
```

jow-, 2015-11-05:
Instead of truncating the hostname we could put it into a <span> container with `max-width`, `white-space: nowrap`, `overflow: hidden` and `text-overflow: ellipsis`.

Dropping noise column would be fine with me, its not available in many drivers anyway.

booo, 2015-11-09:
Please keep the noise column.

thess, 2015-11-09:
I like this but, it would be more useful if this addition could show all associated station's IP and not just the ones using DHCP. A lot my wireless devices have static IP.

jow-, 2015-11-11:
@booo: Why keep it? It serves no useful purpose and you can still find the value by hovering over the signal indicator icon.

jow-, 2015-11-11:
@thess: should be doable by mixing in ARP and NDP cache dumps as well.

booo, 2015-11-11:
@jow- In my day to day activity I compare the signal quite often to the noise. If I have to hover/click each time to see the noise level the interface is far from user friendly.

Maybe someone can post a screenshot or link to the proposed implementation. Maybe the noise level is already a prominent part of the gui and I can't see it in the current screenshot.

jow-, 2015-11-17:
Since the noise floor is global to the entire radio device and not specific to stations or networks it maybe makes sense to display it somewhere in the header. At least it is redundant to repeat it in each row.

booo, 2015-11-18:
I agree that the information is redundant in each row.

echza, 2015-12-05:
how can we enable mauermann's patch without rebuilding the kernel?

FreifunkUFO, 2015-12-21:
it would be nice to have the "expected throughput" in the table..

FreifunkUFO, 2016-01-08:
[image]

if there is 80 MHz wifi bandwidth the table is still showing "20 MHz" :-(

valentt, 2016-01-08:
@FreifunkUFO  I'm seeing 20Mhz when client is resting but when I started download LUCI correctly showed that client switched to 40Mhz channel width.

valentt, 2016-01-08:
@FreifunkUFO How would you propose to know what is "expected throughput" ?!? Is there some forumula for calculating "expected throughput" ?

hnyman, 2016-01-08:
@FreifunkUFO @valentt 
The >40 MHz channel width display is problematic. Apparently there is currently no easy way to pass the 80/160 width info.

Based on https://dev.openwrt.org/ticket/20840 I raised that issue as https://github.com/openwrt/luci/issues/533 , and @jow- answered that fixing that requires patching libiwinfo etc., not just Luci.

valentt, 2016-01-08:
As I openned a https://github.com/openwrt/luci/issues/540 as duplicate ticket, I'll repeat it here, but it is almost identical from OP text:

LUCI  'Associated Stations' list is missing some crucial info, especially for dual band radios.

Currently 'Associated Stations' list is now showing on which radio is client connected, which is really important info to show.

Also showing IP address from ARP or DHCP for each station would be really important info, as should hostname also.

Network column could/should be replaced by Radio name, or should incorporate radio name.

It is common to name both 2.4 and 5.8 network with same SSID if it is on same router, and here LUCI is not usable becaues it doesn't show on which radio which client is connected to.

valentt, 2016-01-08:
Is it possible to arrange info in two rows? I did a quick mock-up to show this idea.
[image]

valentt, 2016-01-08:
Also Gargoyle firmware has some station list done quite nicely so we can borrow ideas from them also.
[image]

valentt, 2016-01-08:
Also there are quite a few online admin dashboard templates that we can look for, find good examples and take some elements from them that make sense for LUCI and OpenWrt.
First example: http://rubix305.sketchpixy.com/app/dashboard

FreifunkUFO, 2016-01-08:
hnyman: ok
valentt: maybe we should use different background colors for these rows, depending on what interface they are? (your problem is only appearing when using same wifi-bssids for different wifi-radios)

about other dashboards: have a look for that brandnew freifunk-gluon status-page https://www.youtube.com/watch?v=lwCwd--Y5R4

Unfortunately that seems to be programmed exclusively for gluon-openwrt (only with centralized "gluon" vpn servers, only with ipv6, you have to use batman-adv and gluon-announce-daemon other needs) 
its only used for getting wifi ad-hoc-statistics but could also be used for Accesspoint-Mode.

psyborg55, 2016-01-08:
not only OP dropped noise column but SSID column too. both columns should stay while there is still enough space left to add phy/radio column. hostname you can see under DHCP leases and expected throughput is not useful as it's incorrect ->https://github.com/openwrt/mt76/issues/17

jow-, 2016-01-08:
The expected throughput is incorrect with the mt76 driver but works fine on all others.

jow-, 2016-01-08:
Disregard the comment above, I thought you meant the phy rate with expected throughput.

raz123, 2016-01-23:
Wouldn't the best option be to allow one to customize which columns to display?

bit2016, 2016-01-28:
+1 I think this is an important feature missing. I filter the MACs and its hard to keep track of devices only displaying their MACs

jow-, 2016-01-28:
I already reworked the assoc list display in 2588364, it looks like this currently:

[image]

psyborg55, 2016-01-29:
so how to build latest trunk without this change?

jow-, 2016-01-29:
Sigh, whats wrong about this change now?

psyborg55, 2016-01-29:
it looks ugly with openwrt.org theme

juanriccio, 2016-06-01:
Recent versions of LuCI do show the hostname and IP address of associated stations, too (nice!), but only when the router is the dhcp provider for that address. I have several routers in AP mode - only the main router has a dhcp server running, so I don't get to see the hostname or IP address.

Could it be possible to query the default (LAN) DNS/DHCP server for the hostname? I guess the IP address could be gathered from local arp tables. If it's too heavy computationally, name lookup could be made optional. I, for one, would welcome that information. It enables me to see at a glance which wireless devices are hooked up to which of the several working routers/APs.

LipkeGu, 2016-06-10:
@juanriccio you can walkaround this via WINS (samba) slave routers shlould use your main router as WINS Master,

bit2016, 2016-06-10:
I think it looks great!

But we should be able to manually assign names to the IPs/clients.

That was my first idea.

juanriccio, 2016-06-10:
@LipkeGu thanks for the suggestion! I set up the main router as a WINS master as detailed for example here: https://wiki.openwrt.org/doc/howto/cifs.server - and I think the setup was successful.

However, I have no idea how to setup the slave routers: they have no Samba service compiled in. That's to keep the image small, I think Samba wouldn't fit. They aren't DHCP clients, either, since they have static IP addresses. How can I set them up to use the master router as a WINS server? Is there any small package that could help? Thank you again.

braian87b, 2016-09-02:
I think that is better to keep aside the macaddress to "name" translation, data can be retrieved from ethers (dnsmasq dhcp file) from other dns, from wins, or even a custom csv, etc, not neccesary needs to be tied to computer hostname or local dns name... could be a friendly name or device user too.

juanriccio, 2016-09-02:
Well it would all be fine, as long as I can spot at a glance which wireless devices are connected to that AP. The MAC address alone is not enough for me; I need some symbolic name, or at least an IP address. However, as I said, IP addresses are only shown when the AP is acting as a DHCP server, which in my network just doesn't happen: I have only the main router serving DHCP and DNS. All the APs are just acting as "dumb" APs.

I've had no success with WINS, either, but maybe I'm doing something wrong.

FreifunkUFO, 2016-09-03:
@juanriccio thats not possible by design (tcp/ip)
the AP has to have some IP interaction with the client to know his IP (but thats not needed for "normal" AP behaviour). it would be possible for an AP without DHCP, f.i. when your AP is in the same IP-Subnet and f.i. ping your client (or whole subnet :-o ) 
in my screenshot the AP is also not the DHCP server (wlan0-2) but somehow AP knows the ipv6 adress of that client..

@jow- would be very nice to have "expected throughput" on the table, even when not working on mt76, would be nice for all others.

@jow- at least the RX/TX rate might be nice when formatting left (not centered!)

[image]

FreifunkUFO, 2016-09-03:
@juanriccio when doing wifi-mesh there was also that problem.. so 9 years ago an extra plugin was written for the routingprotocol olsr, f.i. see https://github.com/servalproject/olsr/blob/master/lib/arprefresh/README_ARPREFRESH

its enabled on my host but in line 2 on the screenshot that doesnt work. but that might be a problem of that plugin or with my meshsetup..

juanriccio, 2016-09-04:
@FreifunkUFO Thanks for the insight about IP interaction. However, some of my APs do show the IPv6 addresses of associated stations, even if I've gone to great lengths to ensure IPv6 is disabled. Mh, mysteries.

My point is: I do keep names for all station, and the main router has all the naming info I need. The info is available via its DNS/DHCP server, which in turns gets it from /etc/ethers, /etc/hosts and the other usual sources. MACaddr/IPaddr matching can be done via arp and grep on the main router.

I've read the page of the arprefresh plugin that you pointed me to, but as you say it stopped working, and I don't even use olsr, so it's too far out for me. Is there a simple way - a script or something - that can fetch the MACaddr/IPaddr matching info back to the APs?

braian87b, 2016-09-04:
@juanriccio in order to disable ipv6 assignation on dnsmasq this should be enought:

    uci del dhcp.lan.dhcpv6
    uci del dhcp.lan.ra
    uci commit dhcp; sync
    /etc/init.d/dnsmasq restart

braian87b, 2016-09-04:
in my system:

`/tmp/dhcp.leases` has all dhcp (static or dynamic) leases, comes from default: `dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'`

`/tmp/etc/dnsmasq.conf` read as `dhcp-host=00:11:22:33:44:57,192.168.1.100,pchostname` for static hosts using `uci add dhcp host` or using luci, lease served or not.

`/etc/ethers` for static hosts (this is the same as `dhcp-host` but just from an external file, lease served or not, just when default `dhcp.@dnsmasq[0].readethers='1'` is `'1'`, could be populated using `echo "00:11:22:33:44:57 192.168.1.100" >> /etc/ethers` this has mac to IP or mac to host, in `/etc/hosts` you could have IP <- hostname, when default `dhcp.@dnsmasq[0].dynamicdhcp='1'` is set to '0' then only host leases in ethers file will be served by dhcp (according to documentation), but actually it serves hosts leases from `/tmp/etc/dnsmasq.conf` with `dhcp-host` too.

I have populated the hosts using uci commands manually, this way it acts I have added using luci.

If i need to I will read the information from `/tmp/dhcp.leases` as described here: http://johnbokma.com/mexit/2008/09/03/dhcp-static-ip-dnsmasq.html

juanriccio, 2016-09-05:
@braian87b wrote:

> in my system:
> `/tmp/dhcp.leases` [...]
> `/tmp/etc/dnsmasq.conf`

Thanks, braian87b! I know where they are, and I can parse them with
scripts. My issue is that they are on the _main router_ (MR).
Unfortunately, the luci table we are talking about is on the AP.

I know how to deal with this on the processing side (MR), but it's not
enough, because I need to:

1. Have the AP query MR for that info (IPaddr, name) and display it.
2. Make MR listen for such requests or provide data whenever it changes
with reasonable granularity.
3. Extract the info from several system files.

I suppose:

- [ ] 1. Asking for and displaying the info is probably a job for
luci/lua (I've got no clue how to do this).
- [ ] 2. Are there any services I can piggyback on?

- [ ] 3. Actually write the awk/sed/whatever to extract the relevant
info, assuming you are fed with a MACaddr.

I can deal with Task #3.

juanriccio, 2016-09-05:
As for the side mistery,

@braian87b depicted:

> ![disabling IPv6 in your browser via
luci](https://cloud.githubusercontent.com/assets/6067040/18232056/3d81a17e-729e-11e6-864c-de2b5c901e59.png)

I'd done this already, that's why I found it annoying! :rage:

braian87b, 2016-09-05:
@juanriccio Sorry, I didn't read correctly what you described, I interpreted something else... I actually managed to get info on AP from MR.

braian87b, 2016-09-05:
opkg update
	opkg install luci-mod-rpc # brings luci-lib-json
	opkg install rpcd-mod-file # brings zlib libopenssl libcurl
	opkg install curl

	/etc/init.d/uhttpd restart
	/etc/init.d/rcpd restart
	# if not works, then reboot:
	ubus list | grep file # check, now should list `file`

	# assing permissions (beware this will add full permissions to rcpd access, you should set as you need
	cat<<'EOF' > /usr/share/rpcd/acl.d/superuser.json
	{
	   "superuser": {
	      "description": "Super user access role",
	      "read": {
	         "ubus": {
	           "*": [ "*" ]
	         },
	         "uci": [ "*" ]
	      },
	      "write": {
	         "ubus": {
	          "*": [ "*" ]
	         },
	         "uci": [ "*" ]
	      }
	   }
	}
	EOF

	rm /tmp/test_output.txt
	IP_HOST_MR=192.168.1.1
	PASSWORD=toor
	. /usr/share/libubox/jshn.sh
	JSON=$(curl -s -d '{ "jsonrpc": "2.0", "id": 1, "method": "call", "params": [ "00000000000000000000000000000000", "session", "login", { "username": "root", "password": "'$PASSWORD'"  } ] }' http://$IP_HOST_MR/ubus)
	json_load "$JSON"
	json_select result
	json_select 2
	json_get_var SESION_ID ubus_rpc_session
	JSON=$(curl -s -d '{ "jsonrpc": "2.0", "id": 1, "method": "call", "params": [ "'$SESION_ID'", "file", "read", { "path": "/tmp/dhcp.leases" } ] }' http://$IP_HOST_MR/ubus)
	json_load "$JSON"
	json_select result
	json_select 2
	json_get_var FILE_DATA data
	echo "$FILE_DATA" > /tmp/test_output.txt
	cat /tmp/test_output.txt

Sorry again for misunderstanding...

braian87b, 2016-09-05:
There is a simple way, just using arp and a simple reverse DNS query, but I don't know if will work for you... `arp -a` should always retrieve _all_ the active hosts on the current AP

	root@OpenWrt:~# arp -a
	IP address       HW type     Flags       HW address            Mask     Device
	192.168.1.34     0x1         0x2         ec:08:6b:xx:xx:xx     *        br-lan
	192.168.1.35     0x1         0x2         84:d6:d0:xx:xx:xx     *        br-lan
		
	root@APAluYBraian:~# arp -a | grep 84:d6:d0:xx:xx:xx
	192.168.9.35     0x1         0x2         84:d6:d0:xx:xx:xx     *        br-lan

	root@OpenWrt:~# nslookup 192.168.1.35  192.168.1.1
	Server:    192.168.1.1
	Address 1: 192.168.1.1 mainrouter
	Name:      192.168.1.35
	Address 1: 192.168.1.35 **KindleFire.lan**

	# keeping this in mind, we can do:
	MAC_TO_QUERY=84:d6:d0:xx:xx:xx
	IP_QUERY=$(arp -a | grep $MAC_TO_QUERY | cut -d' ' -f1)
	HOSTNAME_QUERY=$(nslookup `arp -a | grep $MAC_TO_QUERY | cut -d' ' -f1` 127.0.0.1 | tail -n 1 | grep -o '[^ ]*$' | cut -d' ' -f 1)
	echo $IP_QUERY
	echo $HOSTNAME_QUERY

braian87b, 2016-09-05:
I have the CC 15.05.1 version of OpenWrt, I don't know how to install a newer version of luci without break anything (please if any of you know, tell me), but I have tested the _DHCP Leases_ table, if I replace the `/tmp/dhcp.leases` and obviusly it takes the data from there, maybe the _Associated Stations_ table do the same.

juanriccio, 2016-09-05:
@braian87b  Great replies, thanks!

> There is a simple way, just using arp and a simple reverse DNS query, but I don't know if will work for you...  arp -a  should always retrieve all the active hosts on the current AP

On my AP systems, the `arp` command doesn't report the IP address of associated stations, since as @FreifunkUFO wrote earlier, there's no IP interaction at all, and everything happens at Layer 2. Only the MR knows about IP addresses and symbolic names.

I've cooked up a couple of scripts to be run on the MR via the default `ash` shell. Both take a single parameter - the MAC address of an associated station - and return respectively the IP address and the hostname.

mac2ip:

	#!/bin/ash
	# mac2ip - return the IP address of an associated station given its MAC address
	
	mac2ipaddr()
	{
		grep -i $1 /var/dhcp.leases \
		| head -1 \
		| cut -d' ' -f3
	}
	
	if [ "$#" -eq 1 ]; then
		mac2ipaddr $1
	fi

mac2host:

	#!/bin/ash
	# mac2host - return the hostname of an associated station given its MAC address
	
	mac2hostname()
	{
		grep -i $1 /var/dhcp.leases \
		| head -1 \
		| cut -d' ' -f4
	}
	
	if [ "$#" -eq 1 ]; then
		mac2hostname $1
	fi

However, my knowledge of rpc/JSON mechanisms is null. Assuming you have such scripts available and functional on the MR, how would you go about calling them from the AP?

> I have the CC 15.05.1 version of OpenWrt, I don't know how to install a newer version of luci without break anything (please if any of you know, tell me), but I have tested the DHCP Leases table, if [...]

I don't understand - do you mean trouble with patching the rpc/JSON statements into the AP's luci page again? I guess it shouldn't take much to any knowledgeable guy to fix the luci/lua code. Maybe someone will pick up from there...?  :wink:

juanriccio, 2016-09-06:
@braian87b wrote:

> It is not too pretty but it actually works:
> 
> Run this on some AP and point  IP_HOST_MR  and  PASSWORD  vars to your main Router
> 
>   opkg update
>   opkg install luci-mod-rpc # brings luci-lib-json
>   opkg install rpcd-mod-file # brings zlib libopenssl libcurl
>   opkg install curl

How about the main router? I installed the same packages on the MR, but this code fails on the AP.

> ```
>   # assing permissions (beware this will add full permissions to rcpd access, you should set as you need
>   cat<<'EOF' > /usr/share/rpcd/acl.d/superuser.json
>   {
>      "superuser": {
>         "description": "Super user access role",
>         "read": {
>            "ubus": {
>              "*": [ "*" ]
>            },
>            "uci": [ "*" ]
>         },
>         "write": {
>            "ubus": {
>             "*": [ "*" ]
>            },
>            "uci": [ "*" ]
>         }
>      }
>   }
>   EOF
> 
>   rm /tmp/test_output.txt
>   IP_HOST_MR=192.168.1.1
>   PASSWORD=toor
>   . /usr/share/libubox/jshn.sh
>   JSON=$(curl -s -d '{ "jsonrpc": "2.0", "id": 1, "method": "call", "params": [ "00000000000000000000000000000000", "session", "login", { "username": "root", "password": "'$PASSWORD'"  } ] }' http://$IP_HOST_MR/ubus)
>   json_load "$JSON"
>   json_select result
>   json_select 2
>   json_get_var SESION_ID ubus_rpc_session
>   JSON=$(curl -s -d '{ "jsonrpc": "2.0", "id": 1, "method": "call", "params": [ "'$SESION_ID'", "file", "read", { "path": "/tmp/dhcp.leases" } ] }' http://$IP_HOST_MR/ubus)
>   json_load "$JSON"
>   json_select result
>   json_select 2
>   json_get_var FILE_DATA data
>   echo "$FILE_DATA" > /tmp/test_output.txt
>   rm /usr/share/rpcd/acl.d/superuser.json
> ```

Here's what I get:

    # ./getleases.sh
    WARNING: Variable 'result' does not exist or is not an array/object
    WARNING: Variable '2' does not exist or is not an array/object
    
    # wc /tmp/test_output.txt
            1         0         1 /tmp/test_output.txt
    #

braian87b, 2016-09-06:
@juanriccio 
Sorry, yesterday was about 1:30am I didn't express myself well, please read again the edit post: https://github.com/openwrt/luci/issues/540#issuecomment-244657293
I think I was more clear now.

I mean that I have the ChaosCalmer 15.05.1 version of OpenWrt from https://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/ , I don't know how to install a newer version of luci to test the new **Associated Stations** table and see if it works or not.
I don't know noting about luci actually, I use more terminal than luci... but I want to be able to test it too but I don't know how.

Are this new features present on https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/ ?

juanriccio, 2016-09-06:
@braian87b 
> read again the edited post

now I see: it's the other way round! Now I see. I could have figured it out myself. Thanks for straightening it up; I'm going to play with it tomorrow.

> I use more terminal than luci... but I want to be able to test it too but I don't know how.
> Are this new features present on https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/ ?

I think so, at least if you pick a version with luci. Otherwise, you can install luci via `opkg` - as you did for, say, `curl` - and start it by issuing something like
```
opkg update && opkg install luci
/etc/init.d/uhttpd start
```

braian87b, 2016-09-06:
@juanriccio  Yess, well, I will sysupgrade to snapshots/trunk to see if this is there.

juanriccio, 2016-09-14:
I found some time to try and play with this ideas, and I came up with a script to be run on the main router. Here it is.

```
#!/bin/sh

# includes the functions that we need to parse JSON files
. /usr/share/libubox/jshn.sh

case "$1" in
    list)
        echo '{ "mac2host": { "macaddr": "macaddr" }, "mac2ip": { "macaddr": "macaddr"} }'
    ;;
    call)
        case "$2" in
            mac2host)
                # read the argument
                read input
                # Load the argument into the json for retrieval
                json_load "$input"
                json_get_var mac "macaddr"
                hostname=$(grep -i "$mac" /var/dhcp.leases \
                    | head -1 \
                    | cut -d' ' -f4)
                echo "{ \"hostname\":\"$hostname\" }"
            ;;
            mac2ip)
                # read the argument
                read input
                # Load the argument into the json for retrieval
                json_load "$input"
                json_get_var mac "macaddr"
                ipaddr=$(grep -i "$mac" /var/dhcp.leases \
                    | head -1 \
                    | cut -d' ' -f3)
                echo "{ \"ipaddr\" :\"$ipaddr\" }"
            ;;
        esac
    ;;
esac
```
I placed it in the MR's /usr/libexec/rpcd directory, and it does work locally. 
```
# ubus call mactranslate mac2host  '{"macaddr": "00:11:22:33:44:55"}'
{"hostname":"android-123456abcd123456"}
```
How do I setup things on the AP's side? I need to incorporate the rpc call and resulting data into luci. I also need to set minimal permissions.

braian87b, 2016-09-14:
Run this only once on Main Router (to enable rpc calling from dumb AP's):

	opkg update
	opkg install luci-mod-rpc # brings luci-lib-json

	/etc/init.d/uhttpd restart
	/etc/init.d/rcpd restart
	# if not works, then reboot:
	ubus list | grep mac # check, should list `mactranslate` or `mac2host`

This too, its needed assing permissions (beware this will add full permissions to rcpd access, you should set as you need (read documentation to learn how)

	cat<<'EOF' > /usr/share/rpcd/acl.d/superuser.json
	{
	   "superuser": {
	      "description": "Super user access role",
	      "read": {
	         "ubus": {
	           "*": [ "*" ]
	         },
	         "uci": [ "*" ]
	      },
	      "write": {
	         "ubus": {
	          "*": [ "*" ]
	         },
	         "uci": [ "*" ]
	      }
	   }
	}
	EOF

Run this on some AP and point `IP_HOST_MR` and `PASSWORD` vars to your main Router

Run this only once:

	opkg update
	opkg install curl

Copy and paste this to test

	rm /tmp/test_output.txt
	IP_HOST_MR=192.168.1.1
	PASSWORD=toor
	. /usr/share/libubox/jshn.sh
	JSON=$(curl -s -d '{ "jsonrpc": "2.0", "id": 1, "method": "call", "params": [ "00000000000000000000000000000000", "session", "login", { "username": "root", "password": "'$PASSWORD'"  } ] }' http://$IP_HOST_MR/ubus)
	json_load "$JSON"
	json_select result
	json_select 2
	json_get_var SESION_ID ubus_rpc_session
	JSON=$(curl -s -d '{ "jsonrpc": "2.0", "id": 1, "method": "call", "params": [ "'$SESION_ID'", "mactranslate", "mac2host", { "macaddr": "00:11:22:33:44:55" } ] }' http://$IP_HOST_MR/ubus)
	json_load "$JSON"
	json_select result
	json_select 2
	json_get_var FILE_DATA data
	echo "$FILE_DATA" > /tmp/test_output.txt
	cat /tmp/test_output.txt

`/tmp/test_output.txt` contents (should say this if it worked fine):

    {"hostname":"android-123456abcd123456"}

Please let me know about how it worked.

juanriccio, 2016-09-17:
Thanks, @braian87b!
The scripts worked already. My question was mainly about luci integration and about the minimal permission thing. However, I guess I can run this with full permissions on the server (MR) side, so now it's probably just a matter of integrating and packaging.

Here are the scripts. On the MR:

```
#!/bin/sh

# mactranslate - script for master router
# examines dhcp leases
# and returns hostname and IP addr for a given macaddr

# includes the functions that we need to parse JSON files
. /usr/share/libubox/jshn.sh

case "$1" in
    list)
        echo '{ "mac2host": { "macaddr": "macaddr" }, "mac2ip": { "macaddr": "macaddr"}, "mac2full": {"macaddr": "macaddr"} }'
    ;;
    call)
        case "$2" in
            mac2host)
                # read the argument
                read input
                # Load the argument into the json for retrieval
                json_load "$input"
                json_get_var mac "macaddr"
                hostname=$(grep -i "$mac" /var/dhcp.leases \
                    | head -1 \
                    | cut -d' ' -f4)
                echo "{ \"data\":\"$hostname\" }"
            ;;
            mac2ip)
                # read the argument
                read input
                # Load the argument into the json for retrieval
                json_load "$input"
                json_get_var mac "macaddr"
                ipaddr=$(grep -i "$mac" /var/dhcp.leases \
                    | head -1 \
                    | cut -d' ' -f3)
                echo "{ \"data\":\"$ipaddr\" }"
            ;;
            mac2full)
                # read the argument
                read input
                # Load the argument into the json for retrieval
                json_load "$input"
                json_get_var mac "macaddr"
                hostname=$(grep -i "$mac" /var/dhcp.leases \
                    | head -1 \
                    | cut -d' ' -f4)
                ipaddr=$(grep -i "$mac" /var/dhcp.leases \
                    | head -1 \
                    | cut -d' ' -f3)
                echo "{ \"data\":\"$hostname ($ipaddr)\" }"
            ;;
        esac
    ;;
esac
```

On the AP:
```
# getmacinfo.sh - AP side script
# gets hostname and IP addr for a given macaddr from master router

usage=$(cat <<'END_HEREDOC'
getmacinfo.sh [options] <mac_addr>
options:
 -h  return hostname
 -i  return ipaddress
 -a  return "hostname (ipaddress)"
default: -a
END_HEREDOC
)

IP_HOST_MR=192.168.1.1          # the master router IP address
PASSWORD=toor                   # root password on master rouer
METHOD="mac2full"               # default: "hostname (ipaddress)" string

while getopts ":ahi" opt; do
  case $opt in
    a)
      METHOD="mac2full"
      ;;
    h)
      METHOD="mac2host"
      ;;
    i)
      METHOD="mac2ip"
      ;;
    \?)
      echo "$0 invalid option: -$OPTARG" >&2
      ;;
  esac
done
shift $((OPTIND-1))
if [ $# -ne 1 ]; then
        echo "$usage"
        exit 1
fi

. /usr/share/libubox/jshn.sh

JSON=$(curl -s -d '{ "jsonrpc": "2.0", "id": 1, "method": "call", "params": [ "00000000000000000000000000000000", "session", "login", { "username": "root", "password": "'$PASSWORD'"  } ] }' http://$IP_HOST_MR/ubus)
json_load "$JSON"
json_select result
json_select 2
json_get_var SESSION_ID ubus_rpc_session
JSON=$(curl -s -d '{ "jsonrpc": "2.0", "id": 1, "method": "call", "params": [ "'$SESSION_ID'", "mactranslate", "'$METHOD'", { "macaddr": "'$1'" } ] }' http://$IP_HOST_MR/ubus)
json_load "$JSON"
json_select result
json_select 2
json_get_var FILE_DATA data
echo "$FILE_DATA"

```

A sample incorrect call (performed on the AP):
```
# ./getmacinfo.sh
getmacinfo.sh [options] <mac_addr>
options:
 -h  return hostname
 -i  return ipaddress
 -a  return "hostname (ipaddress)"
default: -a

```

A correct call, again on the AP:
```
# ./getmacinfo.sh 00:11:22:33:44:55
android-123456abcd123456 (192.168.1.52)

```

All that's needed now is to get this bit of information in the "Associated Stations" table :smile:

braian87b, 2016-09-18:
This is amazing, I think now we should gain the atention of a experience developer of Luci to integrate this code and make it work... having this should be really easy to get done, may be this should be added to a package and we could get an experience package dev to help us.

It is really common to need to know the actual device when seeing the Web UI, and more than common to have a Main Router and several dumb AP's... should be great too to get on the Main Router info about in which one of many dumb AP's is connected which device... what do you think? ;)

braian87b, 2016-09-18:
I found this, here is more info about ACL's: https://wiki.openwrt.org/doc/techref/ubus#acls

juanriccio, 2016-09-18:
@braian87b wrote:
> should be great too to get on the Main Router info about in which one of many dumb AP's is connected which device... what do you think? ;)

I agree - however, it's not as simple. I can't think of a way to do this efficiently, without lots of querying or automatic, timed notifications. The matter is I don't know the system, in particular all the system hooks that probably are already in place and could make this an easy task.

> I found this, here is more info about ACL's: https://wiki.openwrt.org/doc/techref/ubus#acls

Aha, thanks! I'll read that doc. Hopefully, I'll figure out how the ACL business works and possibly update the scripts accordingly.

cshoredaniel, 2017-01-14:
I think this (MAC -> address translation) would be a useful feature, but getting it work will take some significant time and testing.  One issue is that you have to handle the case where main router is not a version with OpenWrt/LEDE that can communicate the addresses to the the AP's, and on networks without IP addresses (i.e. pure bridges) the AP won't have the information most of the time.

braian87b, 2017-01-18:
Hi, in my opinion its not a problem on "networks without IP addresses / pure bridges"...
actually the several test on scripts that @juanriccio and me performed is more than sufficient to gather the information on dumb AP's and send it to _(or get if from)_ the Main Router _(the one with dhcp)_ and save it somewhere _(we could have a plaintext tab-delimited or sqlite, is more than enough)_ and then in luci check if that plaintext or sqlite is available and then show as an extra column or instead of macaddress (the macaddres could remain in a tooltip) or even put it as a tooltip or even in a separate new page.

I think that what just we need is to create a init.d script (to act as a pseudo daemon and update the info when a new host appears, and put a minimal uci config and then put it all together in a new package and make it available, I can promise to try to get some init.d work and publish here the final files, but as we previusly talked about with @juanriccio some messages ago # **we need someone** who knows both, how to put the info in luci, convert into a package and with access to publish as package in the official repositories.
imagine that would be something like:

**in the Main Router:** 

    opkg update, opkg install luci-macaddr-to-name macaddr-to-name

 _(one for adding luci info page, and one for daemon that updates the tab-delimited or sqlite db)_

**on the Dumb AP's** _(or in the main router too if it acts as AP too)_: 

    opkg update, opkg install macaddr-to-name-ap

What do you think?

cshoredaniel, 2017-01-20:
@braian87b Unfortunately, I think I may have some rain on this optimism:

1) Assumes both 'dumb' AP and main router are both running on versions or LEDE/OpenWrt with package included.
2) Security & stability: how exactly are you transferring these files, ensuring integrity etc.?
3) Really the transfer part probably belongs in packages feed
4) Need someone with time/resources to put in this who has (or can acquire) the necessary skills.

Thoughts?

[47 more replies]

closed 2018-07-03 by jow-