openwrt/luci #547: Luci requires authorization on each page change

misacek007, 2015-11-11:
This is on chaos calmer. 
I have been using luci+lighttpd with ssl happily over two weeks (set based on openwrt wiki) and then for some reason my router restarted. This weird issue occured where I can login to luci but it does not seem to set the authorization key that have been there before (/;stok=XXXX/) and therefore on every page change I need to log in again.

This happens also with uhttpd and no ssl. No messages in logread that would explain anything.

Can you help please?

jow-, 2015-11-11:
Make sure you have no other tab with luci open.

misacek007, 2015-11-11:
I have thought about this and it does not seem to be the problem as the same behaviour occurs in different browser on different computer too. Also incognito mode makes no difference.

jow-, 2015-11-11:
So what LuCI version is this?

misacek007, 2015-11-11:
root@vostrak:~# opkg list-installed luci
luci - git-15.248.30277-3836b45-1

jow-, 2015-11-11:
Cannot reproduce the problem on my end. It appears you're using the 15.05 release? Your report is the first of its kind so it does not appear to be a widespread problem. Unfortunately I do not really have any particular idea. Maybe check the permissions of /tmp on your router.

misacek007, 2015-11-11:
Yep, I'm using chaos calmer (15.05).
/tmp permissions seems to be correct (rwxrwxrwt). 

Would you be able to compare the list of your packages with mine:

root@OpenWrt:~# opkg list-installed 
base-files - 157-r46767
busybox - 1.23.2-1
dnsmasq - 2.73-1
dropbear - 2015.67-1
file - 5.20-1
firewall - 2015-07-27
fstools - 2015-05-24-09027fc86babc3986027a0e677aca1b6999a9e14
glib2 - 2.43.4-1
htop - 1.0.3-1
ip6tables - 1.4.21-1
iptables - 1.4.21-1
jshn - 2015-06-14-d1c66ef1131d14f0ed197b368d03f71b964e45f8
jsonfilter - 2014-06-19-cdc760c58077f44fc40adbbe41e1556a67c1b9a9
kernel - 3.18.20-1-d0d2aab265f2535385cf8d756315d27c
kmod-ip6tables - 3.18.20-1
kmod-ipt-conntrack - 3.18.20-1
kmod-ipt-core - 3.18.20-1
kmod-ipt-nat - 3.18.20-1
kmod-ipv6 - 3.18.20-1
kmod-lib-crc-ccitt - 3.18.20-1
kmod-nf-conntrack - 3.18.20-1
kmod-nf-conntrack6 - 3.18.20-1
kmod-nf-ipt - 3.18.20-1
kmod-nf-ipt6 - 3.18.20-1
kmod-nf-nat - 3.18.20-1
kmod-nf-nathelper - 3.18.20-1
kmod-ppp - 3.18.20-1
kmod-pppoe - 3.18.20-1
kmod-pppox - 3.18.20-1
kmod-slhc - 3.18.20-1
kmod-virtio-balloon - 3.18.20-1
kmod-virtio-net - 3.18.20-1
kmod-virtio-random - 3.18.20-1
lftp - 4.6.0-1
libattr - 20150220-1
libblkid - 2.25.2-4
libblobmsg-json - 2015-06-14-d1c66ef1131d14f0ed197b368d03f71b964e45f8
libc - 0.9.33.2-1
libexpat - 2.1.0-3
libffi - 3.0.13-1
libgcc - 4.8-linaro-1
libip4tc - 1.4.21-1
libip6tc - 1.4.21-1
libiwinfo - 2015-06-01-ade8b1b299cbd5748db1acf80dd3e9f567938371
libiwinfo-lua - 2015-06-01-ade8b1b299cbd5748db1acf80dd3e9f567938371
libjson-c - 0.12-1
libjson-script - 2015-06-14-d1c66ef1131d14f0ed197b368d03f71b964e45f8
liblua - 5.1.5-1
libmagic - 5.20-1
libmount - 2.25.2-4
libncurses - 5.9-2
libnl-tiny - 0.1-4
libopenssl - 1.0.2d-1
libpthread - 0.9.33.2-1
libreadline - 6.3-1
librpc - 2015-04-10-308e9964bfb623773dc0dcc99ef9d18d1551d6ae
libubox - 2015-06-14-d1c66ef1131d14f0ed197b368d03f71b964e45f8
libubus - 2015-05-25-f361bfa5fcb2daadf3b160583ce665024f8d108e
libubus-lua - 2015-05-25-f361bfa5fcb2daadf3b160583ce665024f8d108e
libuci - 2015-04-09.1-1
libuci-lua - 2015-04-09.1-1
libuuid - 2.25.2-4
libxtables - 1.4.21-1
lua - 5.1.5-1
luci - git-15.248.30277-3836b45-1
luci-app-firewall - git-15.245.46763-b179283-1
luci-base - git-15.245.46763-b179283-1
luci-lib-ip - git-15.245.46763-b179283-1
luci-lib-nixio - git-15.245.46763-b179283-1
luci-mod-admin-full - git-15.245.46763-b179283-1
luci-proto-ipv6 - git-15.245.46763-b179283-1
luci-proto-ppp - git-15.245.46763-b179283-1
luci-theme-bootstrap - git-15.245.46763-b179283-1
mc - 4.8.14-1.3
mtd - 21
netifd - 2015-06-08-8795f9ef89626cd658f615c78c6a17e990c0dcaa
odhcp6c - 2015-07-13-024525798c5f6aba3af9b2ef7b3af2f3c14f1db8
odhcpd - 2015-05-21-2ebf6c8216287983779c8ec6597d30893b914a7c
openssh-client - 6.8p1-1
openssh-client-utils - 6.8p1-1
openssh-keygen - 6.8p1-1
openssh-server - 6.8p1-1
openssh-sftp-client - 6.8p1-1
openssl-util - 1.0.2d-1
opkg - 9c97d5ecd795709c8584e972bfdf3aee3a5b846d-7
ppp - 2.4.7-6
ppp-mod-pppoe - 2.4.7-6
procd - 2015-08-16-0da5bf2ff222d1a499172a6e09507388676b5a08
procps - 3.2.8-1
procps-ps - 3.2.8-1
rpcd - 2015-05-17-3d655417ab44d93aad56a6d4a668daf24b127b84
terminfo - 5.9-2
ubox - 2015-07-14-907d046c8929fb74e5a3502a9498198695e62ad8
ubus - 2015-05-25-f361bfa5fcb2daadf3b160583ce665024f8d108e
ubusd - 2015-05-25-f361bfa5fcb2daadf3b160583ce665024f8d108e
uci - 2015-04-09.1-1
uclibcxx - 0.2.4-1
uhttpd - 2015-08-17-f91788b809d9726126e9cf4384fedbbb0c5b8a73
uhttpd-mod-ubus - 2015-08-17-f91788b809d9726126e9cf4384fedbbb0c5b8a73
usign - 2015-05-08-cf8dcdb8a4e874c77f3e9a8e9b643e8c17b19131
vim-full - 7.4-2
vim-help - 7.4-2
vim-runtime - 7.4-2
vsftpd-tls - 3.0.2-4.1
xinetd - 2.3.15-3
zlib - 1.2.8-1

misacek007, 2015-11-11:
Yes, I'm on chaos calmer (15.05). /tmp permissions are correct (rwxrwxrwt).

Do you know how do I test that luci runs correctly (something like: /usr/bin/lua luci.lua, outputs html)

My lighttpd.conf for reference: http://pastebin.com/cbjZh75K

jow-, 2015-11-11:
Maybe lighttpd is not running as root.

misacek007, 2015-11-11:
Found the problem -- you need rpcd to be running or you get this kind of behaviour.

My fault after all :) I hope it will help someone else with the same problem.

closed 2015-11-17 by jow-