openwrt/luci #552: latest changes luci-mod-admin-full - not show other IP, if 2 interfaces to WAN

build000, 2015-11-18:
eq. /etc/config/network:
```
(...)
config interface 'wan'
	option ifname 'eth0.2'
	option proto 'dhcp'
	option macaddr 'xx:xx:xx:xx:xx:xx'
	option peerdns '0'
	option dns '208.67.222.222 208.67.220.220'
	option mtu '9024'

config interface 'wan6'
	option ifname 'eth0.2'
	option proto 'dhcpv6'

config interface 'wan2'
	option ifname 'wwan0'
	option proto 'ncm'
	option device '/dev/cdc-wdm0'
	option pincode '1234'
	option apn 'vpn'
	option username 'vpn'
	option password 'vpn'
	option ipv6 '0'
(...)
```
bad effect can disable wan (wan6) and enable wan2, like this:
```
ifdown wan && ifdown wan6 && ifup wan2
```
see like this:
[image]
I do not see the IP address and all lines that come with it.

It is connected with the recent amendments to luci-mod-admin-full.
It was previously seen on the second IP interface (wan2).

jow-, 2015-11-19:
Pastebin the outputs of `ifstatus wan`, `ifstatus wan2` and `ifstatus wan6`

build000, 2015-11-19:
http://wklej.org/id/1849450/
modem: E3372s-153 (driver ncm)
router: WNDR4300

jow-, 2015-11-19:
So your ubus ifstatus doesn ot contain any ip addresses at all, this looks more like a bug in the ncm protocol handler to me. I agree that it worked before because we queried the IPs directly from the interface there but I want to get away from that as it introduces various issues.

build000, 2015-11-19:
So the suggestion from my hand and ask:
when in the end you will see the address for 4G modems, not to mention the use by Luci - it was the only way I could easily see this address with the number of TX / RX without resorting to the console?

jow-, 2015-11-19:
I'm not saying that I do not want to fix it, just trying to investigate where. Can you provide a complete copy of `ubus call network.interface dump` ? If there are sensitive IPs in it then please replace them with XXX or something but leave the field.

build000, 2015-11-19:
send email...

build000, 2016-04-08:
minimal positive progress (probably latest changes to luci-base):
- visible IP adress properly (eq. wwan0) to /cgi-bin/luci/admin/status
- again not visible IP adress to /cgi-bin/luci/admin/network
;)

jow-, 2018-09-18:
Closing this as this must be solved in OpenWrt

closed 2018-09-18 by jow-